yaqwsx/KiKit #421: assert failed on JLCPCB fab

StereotypicalSquirrel, 2022-09-27:
### Prerequisites

- [X] I have read [FAQ](https://github.com/yaqwsx/KiKit/blob/master/doc/faq.md)
- [X] I use the KiKit at least 1.1.0 (older version are not supported)

### KiKit version

python -m kikit.ui, version 1.1.1

### KiCAD version

6.0.2

### Operating system

Ubuntu 22.04

### Description

I have succesfully panelized a board with kikit and are trying to fab for JLCPCB

command:
python3 -m kikit.ui fab jlcpcb "myboard_panel.kicad_pcb" exportfolder

Result:
./common/project.cpp(101): assert "m_project_name.IsAbsolute()" failed in setProjectFullName().
Segmentation fault (core dumped)

### Steps to Reproduce

python3 -m kikit.ui fab jlcpcb "myboard_panel.kicad_pcb" exportfolder

./common/project.cpp(101): assert "m_project_name.IsAbsolute()" failed in setProjectFullName().
Segmentation fault (core dumped)

yaqwsx, 2022-09-27:
The error comes from the DRC check. If you quickly need to generate the data, please use the option `--no-drc`.

I committed the fix in 251e2dad5b8e8c815c2426eae57b10c49b059155; could you install the upstream version and test if it resolved the problem for you?

StereotypicalSquirrel, 2022-09-27:
Thankyou for the prompt response!
I was able to succesfully export using both the --no-drc option and fix 251e2da

closed 2022-09-27 by StereotypicalSquirrel