yaqwsx/KiKit #423: Feature request: Add layer coupon to panel rail

korken89, 2022-09-30:
### Prerequisites

- [X] I have read the documentation and the proposed feature is not implemented.

### Description

Hi,

One thing I noticed was missing that would be nice to have is to generate a layer order coupon on the rails of an panel.
I made an example here on how it could look (where the numbers are on each layer):
[image]

Thanks for a great tool!

yaqwsx, 2022-10-09:
Seems useful, and we can implement it in the future.

markh-de, 2024-03-01:
This can also be done in a postprocessing script.

I just hacked something together:

```
from kikit.common import fromMm
from pcbnew import F_Cu, B_Cu, In1_Cu
import shapely

def add_layer_window(panel, offset, cell_dim, text_dim, clearance=0, labels='123456789ABCDEFGHJKLMNPR'):
    """
    Creates a layer window to be put onto the panel rails.

    offset: (x, y): window center, distance from top/left if positive, from bottom/right if negative
    cell_dim: (width, height, cell_border_thickness): window cell dimensions
    text_dim: (width, height, text_thickness): text dimensions
    clearance: distance: extra copper clearance around window
    labels: string: layer labels, one character per layer (default: max. 24 layers, alphanum. w/o IOQSXZ)
    """
    minx, miny, maxx, maxy = panel.panelBBox()
    cu_layer_count = panel.board.GetCopperLayerCount()
    window_pos = ((minx if offset[0] > 0 else maxx) + offset[0],
                  (miny if offset[1] > 0 else maxy) + offset[1])
    start_x = window_pos[0] - cell_dim[0] * cu_layer_count / 2
    start_y = window_pos[1] - cell_dim[1] / 2
    end_x   = start_x + cell_dim[0] * cu_layer_count
    end_y   = start_y + cell_dim[1]
    panel.addKeepout(shapely.geometry.box(
        start_x - clearance, start_y - clearance,
        end_x + clearance, end_y + clearance))
    for layer in range(cu_layer_count):
        kicad_layer = F_Cu if layer == 0 else B_Cu if layer == cu_layer_count - 1 else In1_Cu + layer - 1
        cell_start_x = start_x + layer * cell_dim[0]
        add_rectangle(panel, cell_start_x, start_y, cell_start_x + cell_dim[0], end_y, cell_dim[2], kicad_layer)
        panel.addText(labels[layer], (int(cell_start_x + (cell_dim[0]/2)), int(start_y + (cell_dim[1]/2))),
            width=text_dim[0], height=text_dim[1], thickness=text_dim[2], layer=kicad_layer)

def add_rectangle(panel, sx, sy, ex, ey, thickness, layer):
    """Adds a rectangle made out of four single lines."""
    panel.addLine((sx, sy), (sx, ey), thickness, layer)
    panel.addLine((sx, ey), (ex, ey), thickness, layer)
    panel.addLine((ex, ey), (ex, sy), thickness, layer)
    panel.addLine((ex, sy), (sx, sy), thickness, layer)

def kikitPostprocess(panel, arg):
    # Usage example:
    # Create a layer window on the panel rail near the bottom-right corner.
    add_layer_window(panel,
        (fromMm(-27), fromMm(-5)),
        (fromMm(3.5), fromMm(5), fromMm(0.25)),
        (fromMm(2.5), fromMm(2.5), fromMm(0.25)),
        fromMm(0.35))
```

If you're not yet using a post-processing script, you'll need to add the script via
```
--post 'script: scriptname.py;'
```
to your `kikit panelize` call.

On a 4-layer board the result looks like this:
[image]

Hope it's useful.