yaqwsx/KiKit #424: Kicad command prompt error TypeError: 'type' object is not subscriptable

Kralg, 2022-09-30:
### Prerequisites

- [X] I have read [FAQ](https://github.com/yaqwsx/KiKit/blob/master/doc/faq.md)
- [X] I use the KiKit at least 1.1.0 (older version are not supported)

### KiKit version

1.1.1

### KiCAD version

6.0.7

### Operating system

Windows

### Description

I was trying to run the last script from here https://github.com/yaqwsx/KiKit/discussions/242  in Kicad command prompt, but I am getting an error:
C:\Users\User\Documents\KiCad\6.0>Pnlz.py
Fatal Python error: init_import_size: Failed to import the site module
Python runtime state: initialized
Traceback (most recent call last):
  File "C:\Program Files\KiCad\6.0\bin\\lib\site.py", line 73, in <module>
    import os
  File "C:\Program Files\KiCad\6.0\bin\\lib\os.py", line 29, in <module>
    from _collections_abc import _check_methods
  File "C:\Program Files\KiCad\6.0\bin\\lib\_collections_abc.py", line 12, in <module>
    GenericAlias = type(list[int])
TypeError: 'type' object is not subscriptable

C:\Users\User\Documents\KiCad\6.0>python --version
Python 3.9.10

Can you give a hint how to resolve this?

### Steps to Reproduce

I executed the script from the second post from bottom https://github.com/yaqwsx/KiKit/discussions/242 and I am getting error.

yaqwsx, 2022-10-09:
The error basically tells you that stuff from Python standard library cannot be imported, and it is not related to KiKit at all. A possible cause might be mixed virtual environments: https://stackoverflow.com/questions/69007748/fatal-python-error-init-import-size-failed-to-import-the-site-module-in-anacon

yaqwsx, 2022-12-20:
Closing due to inactivity.