yaqwsx/KiKit #427: Panelize: Separate width settings for each frame rail?

gwideman, 2022-10-08:
### Prerequisites

- [X] I have read the documentation and the proposed feature is not implemented.

### Description

In Kikit panelize frame option, is there any chance we could get a separate width settings for each of the four frame rails?

The motivation is for boards that have components like connectors that overhang the edge of the board.  

The board may need to travel one a conveyor through  Pick-and-place, oven, solder machine and so on, where that conveyor may have chains, clips, clamps, brackets that touch or engage with the board and require unpopulated regions on two or more edges.  This usually necessitates having frame rails around the panel.

On one or more sides, where components overhang into the space of the frame rail, that rail may need to be wider to provide clearance between that overhanging connector and the conveyor mechanism.  In that scenario the rail should be at least some minimum width.  In another scenario it should be a precise particular width, for example to work properly with some fixture that perhaps supports the overhanging component during soldering.

In these case, we don't want to have to use that same fat width all around the entire frame.   Hence it would be great to be able to set the width of each rail separately, or to set a default frame width and override it for one or more sides.  

Also, such a feature should work usefully with the mintotalheight, mintotalwidth parameters.  I think this means that if you explicitly specify one rail width, then mintotalxxx should take up any slack on the opposite rail.  

(As a side note, I'm not sure why a user would want to specify that their panel must be larger than X by Y. Instead I view that mintotalxxx is really a way to set the panel to a precise size, but if the user miscalculates boards size plus requested gaps and frame size, then kikit doesn't take responsibility for preventing an unwanted-size output, it just outputs a larger-than-desired panel.)

yaqwsx, 2022-10-09:
The feature seems reasonable and useful. I will add it to my list.

About minimal panel size - some fab houses (e.g., JLC PCB) have requirements for minimal size. So this option can be hidden in `jlcpcb-base.json` and it ensures that you will never create a panel that would be rejected.

gwideman, 2022-10-09:
Thanks for considering the individual rail width feature!

On mintotalheight, mintotalwidth  -- ah I see what you mean for very tiny boards.

gwideman, 2024-12-11:
Any chance we could bump this request up?  Or did it get done and I didn't notice?

This requirement has come up again today, so I thought I'd check whether someone had requested it. Sure enough, someone had... with a name looking very similar to mine. Hahaha.

Anyhow, I'll contribute a suggestion.  The separate width parameters for the four frame rails can be:
- widthR
- widthL
- widthT
- widthB
... and, when provided, they should override the width param.