yaqwsx/KiKit #432: Board with irregular edge cuts does not properly Separate in a multi PCB file.

BillyPocketNC, 2022-10-13:
### Prerequisites

- [X] I have read [FAQ](https://github.com/yaqwsx/KiKit/blob/master/doc/faq.md)
- [X] I use the KiKit at least 1.1.0 (older version are not supported)

### KiKit version

1.1.1

### KiCAD version

6.0.8

### Operating system

Windows 11

### Description

The Edge cut lines are replaced with arcs and are replaced differently depending on where the PCB is located in the file. 
The examples below were chosen randomly to illustrate the inconsistencies.
Original outline 
[image]
Board 1
[image]
Board 4
[image]
Board 6
[image]

Thank you for all your work on this project @yaqwsx .

### Steps to Reproduce

I ran the following in KiCad command prompt under kikit v1.1.1, v1.1.0 and v1.0.5
The problem went away when I ran the commands under v1.0.5

`kikit separate -s "annotation; ref: B1; tolerance: 5mm" BrokenOutline.kicad_pcb panel\board_1.kicad_pcb
kikit separate -s "annotation; ref: B2; tolerance: 5mm" BrokenOutline.kicad_pcb panel\board_2.kicad_pcb
kikit separate -s "annotation; ref: B3; tolerance: 5mm" BrokenOutline.kicad_pcb panel\board_3.kicad_pcb
kikit separate -s "annotation; ref: B4; tolerance: 5mm" BrokenOutline.kicad_pcb panel\board_4.kicad_pcb
kikit separate -s "annotation; ref: B5; tolerance: 5mm" BrokenOutline.kicad_pcb panel\board_5.kicad_pcb
kikit separate -s "annotation; ref: B6; tolerance: 5mm" BrokenOutline.kicad_pcb panel\board_6.kicad_pcb
`
[BrokenOutline.zip](https://github.com/yaqwsx/KiKit/files/9780358/BrokenOutline.zip)

roboteux, 2022-10-20:
I have a similar issue. (kicad 6.0.8 kitkit 1.1.1).
In my case the issue seem related to the presence or arc in the board outline.
the line following an arc is removed from the outline, and the arc goes from it starting point to the end of the missing line

My original board : 
[image]

The board produce after a split
[image]

The kicad file, as well as a .bat calling kitkit
[prop.zip](https://github.com/yaqwsx/KiKit/files/9830866/prop.zip)

kikit separate --source "annotation; tolerance: 25mm; ref: B151" proptest_edge.kicad_pcb prop_main_151.kicad_pcb

roboteux, 2022-10-24:
Hi,
I've retested with Kicad 6.0.7 and the bug is also present.

I've made a simpler test case (attached). I think the bug is only visible with arc that goes counter clockwise.

Also the lines that are missing after the split are still there but with have a zero lenght.

original edge : 
[image]

 (gr_line (start 85.999983 15) (end 17.999983 15) (layer "Edge.Cuts") (width 0.05) (tstamp 00000000-0000-0000-0000-0000606ec39f))
  (gr_arc (start 85.999983 15) (mid 88.121303 15.878675) (end 88.999983 18) (layer "Edge.Cuts") (width 0.05) (tstamp 00000000-0000-0000-0000-000060969731))
  (gr_line (start 88.999983 18) (end 15 17.99) (layer "Edge.Cuts") (width 0.05) (tstamp 00000000-0000-0000-0000-000060e66552))
  (gr_line (start 17.999983 15) (end 15 17.99) (layer "Edge.Cuts") (width 0.05) (tstamp fb5717a1-4a72-4f60-ab05-a7918860908b))

splitted : 
[image]

  (gr_line (start 17.999983 15) (end 17.999983 15) (layer "Edge.Cuts") (width 0.1) (tstamp 38f7b099-ce22-46fb-9316-792398bd91b9))
  (gr_arc (start 15 17.99) (mid 50.227316 8.189547) (end 86.151931 15.003849) (layer "Edge.Cuts") (width 0.1) (tstamp 75a03b78-c744-4620-a88c-8020af555a71))
  (gr_line (start 15 17.99) (end 17.999983 15) (layer "Edge.Cuts") (width 0.1) (tstamp b51a8a65-4530-48a6-8fc5-2b26d2626659))
  (gr_line (start 17.999983 15) (end 86.151931 15.003849) (layer "Edge.Cuts") (width 0.1) (tstamp d3e50a22-4ce0-4e40-824c-7d26e69dccad))

[proptestedge.zip](https://github.com/yaqwsx/KiKit/files/9850688/proptestedge.zip)

yaqwsx, 2022-10-24:
This seems to be the same issue as #414. Since 1.1 release, we enforce arc reconstruction in the separate command, but this feature included a bug that was triggered on some boards. This should be fixed in the upstream version (b7078f92b0e8464b3d28aa84e1b6e42ef1d5ca7f) and I also added switches to the separate command to control whether arcs should be preserved or not (b7078f92b0e8464b3d28aa84e1b6e42ef1d5ca7f).

If the problem preserves, please feel free to reopen the issue.

closed 2022-10-24 by yaqwsx