yaqwsx/KiKit #433: ignore graphics refdes "G"

casundra, 2022-10-13:
### Prerequisites

- [X] I have read the documentation and the proposed feature is not implemented.

### Description

Graphics import into pcbnew with the refdes prefix "G" and I usually don't stop to manually give them a unique number.  There were multiple on my board, and KiKit failed to panelize with an error of not being able to find refdes ' '.  Giving all graphics a unique number solved this.  If ignoring "G" isn't a good idea for other reasons, would be nice to have a more specific error message.

yaqwsx, 2022-10-18:
Could you provide me with an example board and steps to reproduce?

yaqwsx, 2023-07-24:
Closing due to inactivity. If you provide a concrete example and better explanation, feel free to reopen.