yaqwsx/KiKit #441: Number of symbols is not equal to the number of instances

putyn, 2022-10-30:
### Prerequisites

- [X] I have read [FAQ](https://github.com/yaqwsx/KiKit/blob/master/doc/faq.md)
- [X] I use the KiKit at least 1.1.0 (older version are not supported)

### KiKit version

kikit, version 1.1.1

### KiCAD version

6.0.8

### Operating system

windows 10

### Description

Trying to generate assembly files for a project I'm working on, I initially got this error message, not very descriptive
```
An error occurred:
No output files produced
```
The error is raised during the component extraction part, from an assertion where the number of symbols is checked against the number of instances, in case the two are not equal; maybe the assertion can include a message so it would be easier to debug.

### Steps to Reproduce

see project: [epaper-breakout.zip](https://github.com/yaqwsx/KiKit/files/9896723/epaper-breakout.zip)
try to generate assembly data for jlcpcb
kikit fab jlcpcb --drc --assembly --schematic epaper-breakout.kicad_sch epaper-breakout.kicad_pcb .\output

yaqwsx, 2022-10-31:
Hi, is it possible that the file `tps65185x.kicad_sch` used to be a top-level schematic file and was copied inside this project? Strangely enough, this file contains symbol instances, however it is not a top-level file. This triggers an assert (which is reported in  a wrong way) but before I implement any fix, I need to know how it is possible that non top-level file (with no subsheets) contains symbol instances.

putyn, 2022-10-31:
thanks for taking a look at the issue.
There is a high change the tps65185x.kicad_sch was actually a top-level schematic file and after used as a hierarchical sheet; project started a some months ago without a clear structure.

yaqwsx, 2022-10-31:
This should be fixed in 933cbf9027fda067fb094539856e414cf06e912a

closed 2022-10-31 by yaqwsx