yaqwsx/KiKit #453: Separate fails to copy outline if it contains circles with angles other then multiple of 90 degrees

hvraven, 2022-12-06:
### Prerequisites

- [X] I have read [FAQ](https://github.com/yaqwsx/KiKit/blob/master/doc/faq.md)
- [X] I use the KiKit at least 1.1.0 (older version are not supported)

### KiKit version

kikit, version 1.1.2

### KiCAD version

6.0.7

### Operating system

Linux, voidlinux

### Description

`kikit separate` fails to correctly copy the board outline when it contains circles which do not end at 90 degree angles (at least thats what I have narrowed it down to).

Input:
[image]
Output:
[image]

Project repository with input & output: [mwe.zip](https://github.com/yaqwsx/KiKit/files/10166470/mwe.zip)

### Steps to Reproduce

* Create a board with outline with circles which do not start & end at 90 degrees intervals
* Run `kikit separate -s 'annotation; ref: B1' test.kicad_pcb out.kicad_pcb`

yaqwsx, 2022-12-06:
This is a bug in the arcs reconstruction method. As a workaround before we fix this, you can add `--looseArcs` to your command.

hvraven, 2022-12-06:
Thank you very much for the quick response. This does indeed solve the issue and generate the expected output.

yaqwsx, 2024-03-06:
As the switch to a new geometry engine will take some time, I tried to fix the current arc welder in a2c9891. The upstream version should fix your issue.

yaqwsx, 2024-03-15:
I tested the arc welder on a dozen of designs, seems to be working fine.

closed 2024-03-15 by yaqwsx