yaqwsx/KiKit #455: ModuleNotFoundError: No module named 'pcbnew',

martin323232, 2022-12-15:
### Prerequisites

- [X] I have read [FAQ](https://github.com/yaqwsx/KiKit/blob/master/doc/faq.md)
- [X] I use the KiKit at least 1.1.0 (older version are not supported)

### KiKit version

pip install kikit

### KiCAD version

6.0.9

### Operating system

Windows 10

### Description

KiKit worked under kicad 6.0.7 after upgrading to 6.0.9 got message that backend was missing.
Installed latest version of KiKit using pip install, using "kikit --help" gives output below.
Python version 3.10.1

`Traceback (most recent call last):
  File "C:\Users\m_mau\AppData\Local\Programs\Python\Python310\lib\runpy.py", line 196, in _run_module_as_main
    return _run_code(code, main_globals, None,
  File "C:\Users\m_mau\AppData\Local\Programs\Python\Python310\lib\runpy.py", line 86, in _run_code
    exec(code, run_globals)
  File "C:\Users\m_mau\AppData\Local\Programs\Python\Python310\Scripts\kikit.exe\__main__.py", line 4, in <module>
  File "C:\Users\m_mau\AppData\Local\Programs\Python\Python310\lib\site-packages\kikit\ui.py", line 2, in <module>
    from kikit import (panelize_ui, export_ui, present_ui, stencil_ui,
  File "C:\Users\m_mau\AppData\Local\Programs\Python\Python310\lib\site-packages\kikit\panelize_ui.py", line 7, in <module>
    from kikit.panelize_ui_sections import *
  File "C:\Users\m_mau\AppData\Local\Programs\Python\Python310\lib\site-packages\kikit\panelize_ui_sections.py", line 4, in <module>
    from kikit import plugin
  File "C:\Users\m_mau\AppData\Local\Programs\Python\Python310\lib\site-packages\kikit\plugin.py", line 10, in <module>
    from kikit.panelize import Panel
  File "C:\Users\m_mau\AppData\Local\Programs\Python\Python310\lib\site-packages\kikit\panelize.py", line 3, in <module>
    from pcbnewTransition import pcbnew, isV6
  File "C:\Users\m_mau\AppData\Local\Programs\Python\Python310\lib\site-packages\pcbnewTransition\__init__.py", line 6, in <module>
    from .transition import *
  File "C:\Users\m_mau\AppData\Local\Programs\Python\Python310\lib\site-packages\pcbnewTransition\transition.py", line 2, in <module>
    import pcbnew
ModuleNotFoundError: No module named 'pcbnew'`

output from install

`My Documents\KiCad\6.0\scripting\plugins>pip.exe install kikit
Requirement already satisfied: kikit in c:\users\m_mau\appdata\local\programs\python\python310\lib\site-packages (1.1.2)
Requirement already satisfied: numpy in c:\users\m_mau\appdata\local\programs\python\python310\lib\site-packages (from kikit) (1.23.5)
Requirement already satisfied: solidpython>=1.1.2 in c:\users\m_mau\appdata\local\programs\python\python310\lib\site-packages (from kikit) (1.1.3)
Requirement already satisfied: click>=7.1 in c:\users\m_mau\appdata\local\programs\python\python310\lib\site-packages (from kikit) (8.1.3)
Requirement already satisfied: shapely>=1.7 in c:\users\m_mau\appdata\local\programs\python\python310\lib\site-packages (from kikit) (2.0.0)
Requirement already satisfied: markdown2>=2.4 in c:\users\m_mau\appdata\local\programs\python\python310\lib\site-packages (from kikit) (2.4.6)
Requirement already satisfied: pcbnewTransition==0.2.0 in c:\users\m_mau\appdata\local\programs\python\python310\lib\site-packages (from kikit) (0.2.0)
Requirement already satisfied: pybars3>=0.9 in c:\users\m_mau\appdata\local\programs\python\python310\lib\site-packages (from kikit) (0.9.7)
Requirement already satisfied: commentjson>=0.9 in c:\users\m_mau\appdata\local\programs\python\python310\lib\site-packages (from kikit) (0.9.0)
Requirement already satisfied: colorama in c:\users\m_mau\appdata\local\programs\python\python310\lib\site-packages (from click>=7.1->kikit) (0.4.6)
Requirement already satisfied: lark-parser<0.8.0,>=0.7.1 in c:\users\m_mau\appdata\local\programs\python\python310\lib\site-packages (from commentjson>=0.9->kikit) (0.7.8)
Requirement already satisfied: PyMeta3>=0.5.1 in c:\users\m_mau\appdata\local\programs\python\python310\lib\site-packages (from pybars3>=0.9->kikit) (0.5.1)
Requirement already satisfied: pypng<0.0.20,>=0.0.19 in c:\users\m_mau\appdata\local\programs\python\python310\lib\site-packages (from solidpython>=1.1.2->kikit) (0.0.19)
Requirement already satisfied: ply<4.0,>=3.11 in c:\users\m_mau\appdata\local\programs\python\python310\lib\site-packages (from solidpython>=1.1.2->kikit) (3.11)
Requirement already satisfied: euclid3<0.2.0,>=0.1.0 in c:\users\m_mau\appdata\local\programs\python\python310\lib\site-packages (from solidpython>=1.1.2->kikit) (0.1)
Requirement already satisfied: PrettyTable==0.7.2 in c:\users\m_mau\appdata\local\programs\python\python310\lib\site-packages (from solidpython>=1.1.2->kikit) (0.7.2)`

Is there a path error?

### Steps to Reproduce

Install kikit

yaqwsx, 2022-12-15:
TLDR: You have to install Kikit in KiCAD's command prompt, not the regular command prompt.

KiCAD comes with a bundled Python interpreter that knows the `pcbnew` module. To install and run KiKit you have to use this interpreter (which is by-default available in the KiCAD command prompt; this is the reason why it is provided). Based on the logs, the installation was done in Python interpreter installed by you. Which, doesn't know the `pcbnew` module.

martin323232, 2022-12-15:
sorry about that, fixed the issue. 
Thanks for your help.

closed 2022-12-15 by martin323232