yaqwsx/KiKit #463: [Separate] Output board not centered as the docs states

set-soft, 2022-12-28:
### Prerequisites

- [X] I have read [FAQ](https://github.com/yaqwsx/KiKit/blob/master/doc/faq.md)
- [X] I use the KiKit at least 1.1.0 (older version are not supported)

### KiKit version

1.1.2

### KiCAD version

6.0.9

### Operating system

Linux

### Description

When using `kikit separate` I always get the PCB in the top left corner, not at the center of the page.

Also: the code has a *hardcoded center* (150, 100), not the center of the page.

### Steps to Reproduce

kikit separate -s "annotation; ref: B1" multiboard.kicad_pcb test.kicad_pcb

yaqwsx, 2022-12-28:
Thanks for the report. Could you point me to where the documentation claims the separated board should be centered in the list? I struggle to find it.

The board is positioned according to the options in the `--page` settings (see https://github.com/yaqwsx/KiKit/blob/master/doc/panelizeCli.md#page for more details). There is no hard-coded position [150, 100] - that's just a temporary position that, on purpose, uses these arbitrary numbers to simplify position tracing during debugging, but it does not affect the final position of the panel (panel is positioned in `positionPanel`).

At the moment, the way to center the board is to set the anchor to `c` and `posx` and `posy` to half of the selected page size.

PS: I think that we should consider adding percentage values for `posx` and `posy` of the page size. With that, `c` and `50%` could be the defaults.

yaqwsx, 2022-12-28:
I liked the idea of having the panel placed in the center of the page that I implemented it in 794b7d9. I decided to center only the X-position as centering the Y-position yields interference with the page stamp.

set-soft, 2023-01-03:
Hi @yaqwsx !

Is for the separate subcommand, not for panelize. Running KiKit with separate you get:
```
$ kikit separate --help
Usage: kikit separate [OPTIONS] INPUT OUTPUT

  Separate a single board out of a multi-board design. The separated board
  is placed in the middle of the sheet.

  You can specify the board via bounding box or annotation. See
  documentation for further details on usage.

Options:
  -s, --source PARAMETER_LIST     Specify source settings.
  -P, --page PARAMETER_LIST       Override page settings.
  --debug PARAMETER_LIST          Include debug traces or drawings in the
                                  panel.

  --keepAnnotations / --stripAnnotations
                                  Do not strip annotations
  --preserveArcs / --looseArcs    Preserve arcs in the files
  --help                          Show this message and exit.
```

Where it says *The separated board is placed in the middle of the sheet.*
This is from: https://github.com/yaqwsx/KiKit/blob/e70708034f3fc9cb454ff0423964fce27bd0333a/kikit/panelize_ui.py#L320-L321

And then the code does:
https://github.com/yaqwsx/KiKit/blob/e70708034f3fc9cb454ff0423964fce27bd0333a/kikit/panelize_ui.py#L347-L348

yaqwsx, 2023-01-03:
Oh, it's in the docstring. Thanks. 

> And then the code does:
> 
> https://github.com/yaqwsx/KiKit/blob/e70708034f3fc9cb454ff0423964fce27bd0333a/kikit/panelize_ui.py#L347-L348

As I said, that is not relevant, this is relevant:

https://github.com/yaqwsx/KiKit/blob/e70708034f3fc9cb454ff0423964fce27bd0333a/kikit/panelize_ui.py#L352

Nevertheless, this is fixed in master, closing.

closed 2023-01-03 by yaqwsx