yaqwsx/KiKit #464: Kitkit with Docker on Apple M1 Pro

BenRoe, 2023-01-11:
### Prerequisites

- [X] I have read [FAQ](https://github.com/yaqwsx/KiKit/blob/master/doc/faq.md)
- [X] I use the KiKit at least 1.1.0 (older version are not supported)

### KiKit version

nightly-m1

### KiCAD version

6.0.9

### Operating system

Linux

### Description

I try to use Kitkit with docker on a Apple M1 Pro (13.1).
I get the following error. 
`WARNING: The requested image's platform (linux/arm64v8) does not match the detected host platform (linux/arm64/v8) and no specific platform was requested`

### Steps to Reproduce

I open the the docker container with the command from [here](https://github.com/yaqwsx/KiKit/blob/master/doc/installation.md#running-kikit-via-docker).

`docker run -it -w /kikit -v $(pwd):/kikit yaqwsx/kikit:nightly-m1 /bin/bash`

yaqwsx, 2023-01-11:
Try pulling the container now, please.

BenRoe, 2023-01-11:
This resolves the error.
But now i get this error inside of the container.

Tried to install 

```
Traceback (most recent call last):
  File "/usr/local/bin/kikit", line 11, in <module>
    load_entry_point('KiKit==1.2.0+6.g877475b', 'console_scripts', 'kikit')()
  File "/usr/lib/python3/dist-packages/pkg_resources/__init__.py", line 490, in load_entry_point
    return get_distribution(dist).load_entry_point(group, name)
  File "/usr/lib/python3/dist-packages/pkg_resources/__init__.py", line 2854, in load_entry_point
    return ep.load()
  File "/usr/lib/python3/dist-packages/pkg_resources/__init__.py", line 2445, in load
    return self.resolve()
  File "/usr/lib/python3/dist-packages/pkg_resources/__init__.py", line 2451, in resolve
    module = __import__(self.module_name, fromlist=['__name__'], level=0)
  File "/usr/local/lib/python3.8/dist-packages/KiKit-1.2.0+6.g877475b-py3.8.egg/kikit/ui.py", line 2, in <module>
    from kikit import (panelize_ui, export_ui, present_ui, stencil_ui,
  File "/usr/local/lib/python3.8/dist-packages/KiKit-1.2.0+6.g877475b-py3.8.egg/kikit/panelize_ui.py", line 7, in <module>
    from kikit.panelize_ui_sections import *
  File "/usr/local/lib/python3.8/dist-packages/KiKit-1.2.0+6.g877475b-py3.8.egg/kikit/panelize_ui_sections.py", line 4, in <module>
    from kikit import plugin
  File "/usr/local/lib/python3.8/dist-packages/KiKit-1.2.0+6.g877475b-py3.8.egg/kikit/plugin.py", line 10, in <module>
    from kikit.panelize import Panel
  File "/usr/local/lib/python3.8/dist-packages/KiKit-1.2.0+6.g877475b-py3.8.egg/kikit/panelize.py", line 5, in <module>
    from kikit.common import normalize
  File "/usr/local/lib/python3.8/dist-packages/KiKit-1.2.0+6.g877475b-py3.8.egg/kikit/common.py", line 11, in <module>
    from shapely.geometry import LinearRing
  File "/usr/local/lib/python3.8/dist-packages/shapely/geometry/__init__.py", line 4, in <module>
    from .base import CAP_STYLE, JOIN_STYLE
  File "/usr/local/lib/python3.8/dist-packages/shapely/geometry/base.py", line 20, in <module>
    from shapely.coords import CoordinateSequence
  File "/usr/local/lib/python3.8/dist-packages/shapely/coords.py", line 10, in <module>
    from shapely.geos import lgeos
  File "/usr/local/lib/python3.8/dist-packages/shapely/geos.py", line 100, in <module>
    _lgeos = load_dll('geos_c', fallbacks=alt_paths)
  File "/usr/local/lib/python3.8/dist-packages/shapely/geos.py", line 54, in load_dll
    raise OSError(
OSError: Could not find lib geos_c or load any of its variants ['libgeos_c.so.1', 'libgeos_c.so'].
```

BenRoe, 2023-01-11:
Thanks to @Conor-Burns
```bash
apt install libgeos-dev
```
... resolves the problem.

yaqwsx, 2023-01-11:
The `-dev` version of the library shouldn't be needed, and it is actually rather unwanted. Isn't `libgeos` sufficient? It is strange, though, that the python package `shapely` doesn't bring it in.

BenRoe, 2023-01-11:
If i try `apt install libgeos` i get the error `Unable to locate package libgeos`.
With `libgeos-dev` it works.
Maybe you find an other solution.

yaqwsx, 2023-01-12:
That seems to really be the case. I, unfortunately, don't have easy access to Mac M1, so I for the moment, I am OK with the proposed solution. Thank you for the PR.

commit bda7fe39 (2023-01-12): Add additional packages for the docker M1 image

- add libgeos-dev to fix #464
closed 2023-01-12 by yaqwsx