yaqwsx/KiKit #470: Installing the libraries using the Plugin and Content Manager doesn't work as expected

j4m3s, 2023-01-11:
### Prerequisites

- [X] I have read [FAQ](https://github.com/yaqwsx/KiKit/blob/master/doc/faq.md)
- [X] I use the KiKit at least 1.1.0 (older version are not supported)

### KiKit version

kikit, version 1.2.0

### KiCAD version

6.0.10-86aedd382b~118~ubuntu20.04.1, release build

### Operating system

Ubuntu 20.04

### Description

I have installed the kikit libraries using the Plugin and Content Manager, but they don't show up in either the symbol library or the footprint library.  

I have manually tracked down the `kikit.kicad_sym` file under `${KICAD6_3RD_PARTY}/symbols/com_github_yaqwsx_kikit-library/kikit.kicad_sym` and manually added it via Preferences > Manage Symbol Libraries, but I wouldn't have thought that step should be necessary should it?  If it is:
a) I'll submit a PR to add it to the installation docs
b) I'm wondering what the point of the Content Manager in Kicad is, it's only done half the job and left me to do the legwork :)

### Steps to Reproduce

1. Open the "Plugin and Content Manager"
2. Select the "Libraries" tab
3. Scroll down and click "Install"
4. Press "Apply Changes"

Optionally, restart KiCad.  It doesn't help :(

yaqwsx, 2023-01-12:
Hi! This is mainly a problem in KiCAD please, open an issue on the KiCAD issue tracker. If you open a PR with documentation improvement, I will be happy to merge it.

j4m3s, 2023-01-13:
Have added the issue to gitlab [here](https://gitlab.com/kicad/addons/repository/-/issues/5).  Will create the PR to update the kikit docs :)

j4m3s, 2023-01-14:
The KiCad maintainers have confirmed the install via PCM will work correctly in v7 :)  I'll update my PR to reflect that this relates to v6 - I expect it'll be a while before everyone moves over to 7.

j4m3s, 2023-01-26:
Hi @yaqwsx

Are you OK with this PR or would you like some changes? It's purely a documentation update to help others get the plug in working.

PR #472 (2023-02-01): Add installation instructions for the symbols & footprints via PCM
Fixes #470
commit e0b78b1b (2023-02-01): Add installation instructions for the symbols & footprints via PCM

Fixes #470
closed 2023-02-01 by yaqwsx