yaqwsx/KiKit #473: Error occurred: Discontinuous outline

Razer41, 2023-01-14:
### Prerequisites

- [X] I have read [FAQ](https://github.com/yaqwsx/KiKit/blob/master/doc/faq.md)
- [X] I use the KiKit at least 1.1.0 (older version are not supported)

### KiKit version

1.2.0

### KiCAD version

6.0.10

### Operating system

Windows

### Description

I usually manually panelize PCBs with KiCAD but recently heard about KiKit. 
I'm having an issue with panelizing this pcb in particular and getting the error "Discontinuous outline" at multiple points when attempting to panelize. 
The edge cuts appear near perfect to me so I'm not sure what to change to get this to work. Any help would be greatly appreciated.

### Steps to Reproduce

kikit panelize^
    --layout "rows: 3; cols: 4" ^
    --source "tolerance: 10mm" ^
    --tabs "type: full" ^
    --cuts "type: vcuts; clearance: 0.5mm" ^
    --framing "type: frame; hspace: 1mm; vspace: 1mm; width: 10mm" ^
    C:\Users\x\Downloads\pcb\pcb\KiCad\pcb.kicad_pcb panel.kicad_pcb

PCB: https://drive.google.com/file/d/14q7Kti2q4E7ClnPnHKV7SYEKpfzqB3MV/view?usp=sharing

yaqwsx, 2023-01-14:
Are you sure that the PCB that you provided doesn't work? I tested it in upstream and v1.2.0, and it works just fine. Is it possible that you fixed the misalignment of Edges and just forgot to save the PCB before panelizing?

Razer41, 2023-01-14:
Yes, I just tried with the exact file I posted. Here's what happens:

https://imgur.com/vWMgWXF

I'm using the GUI and installed KiKit with the plugin & content manager and ran "pip3 install kikit" via kicad 6.0 cmd

yaqwsx, 2023-01-14:
But that is a completely different error message than you claim in the issue's title! Please, when reaching out for help, make sure that you provide accurate information.

I understand why this error message that says the cuts are not straight and thus cannot be performed by VCuts is confusing. As the design shouldn't yield any curved cuts. There was indeed a numerical rounding bug, that included also a small portion of the PCB fillet into the cut. This was fixed in [4b96b7c](https://github.com/yaqwsx/KiKit/commit/4b96b7ca79e0cd7f0888c0345c615446008f07fa). 

PS: I also added a new option for full tabs `patchcorners` that should be set to false when you have a design with fillets (I haven't penalized design with fillets with full tabs, so I haen't noted that is actually useful).

closed 2023-01-14 by yaqwsx

Razer41, 2023-01-14:
Apologies, I'm not sure what happened there with the error.

How am I able to disable 'patchcorners'? The tabs setting is certainly the issue as I'm now able to get 'panelize' to work, but only when 'tabs' is set to 'type:none' which of course is not useable as there are no v-scores between each PCB.
Picture: https://imgur.com/xg0WaLY

Thanks for being patient with me, am new to this.

yaqwsx, 2023-01-15:
The bug fixes I talked about and the new feature with patchcorner are not released yet; they were implemented only 11 hours ago. If you want to use them, you have to use upstream version of KiKit (https://github.com/yaqwsx/KiKit/blob/master/doc/installation.md#installing-a-special-version-of-kikit)