yaqwsx/KiKit #480: Mousebite holes are duplicated when space is 0

j4m3s, 2023-01-27:
### Prerequisites

- [X] I have read [FAQ](https://github.com/yaqwsx/KiKit/blob/master/doc/faq.md)
- [X] I use the KiKit at least 1.1.0 (older version are not supported)

### KiKit version

kikit, version 1.2.0

### KiCAD version

6.0.10-86aedd382b~118~ubuntu20.04.1, release build

### Operating system

Ubuntu 20.04

### Description

Actual:  When panelising a board with full tabs and no space and no offset, two rows of mousebites are created per "break" between boards, one superimposed directly on top of the other.  This causes one DRC error for every mousebite drill hole, as there are actually two ("Error: Drilled holes too close together (board setup constraints min 0.2495mm; actual 0.0000mm").

Expected: A single row of mousebite drill holes should be created per board boundary.

### Steps to Reproduce

```json
{
  "layout": {
    "type": "grid",
    "rows": 2,
    "cols": 1
  },
  "tabs": {
    "type": "full"
  },
  "cuts": {
    "type": "mousebites"
  }
}
```
Then open the kicad_pcb file and run the DRC check (or click on a mousebite drill hole and Kicad will ask which of the two superimposed holes you want to select).

yaqwsx, 2023-02-20:
Implemented in 125c752. Originally, I haven't thought about someone using mousebites without tabs, but I guess it makes sense when your boards have some narrow features that can serve as tabs.

closed 2023-02-20 by yaqwsx

j4m3s, 2023-02-20:
Exactly - I have narrow boards and only need to panelise lengthways kif
that makes sense).

Thank you!