yaqwsx/KiKit #483: Adding KIKIT from the Plugin and Content Manager brakes IBOM

dunk-law, 2023-01-27:
### Prerequisites

- [X] I have read [FAQ](https://github.com/yaqwsx/KiKit/blob/master/doc/faq.md)
- [X] I use the KiKit at least 1.1.0 (older version are not supported)

### KiKit version

1.2.0

### KiCAD version

6.0.10

### Operating system

Windows

### Description

If I install the KIKIT application interface, IBOM ceases to function. If I remove KIKIT application interface, the IBOM starts to function again.

### Steps to Reproduce

Report from IBOM with KIKIT interface installed. 
[image]

yaqwsx, 2023-01-27:
The problem comes from https://github.com/yaqwsx/pcbnewTransition. I believe that you must be using a newer KiKit than 1.2.0 (you have to be using the upstream version). 

Pcbnew transition monkey patches the Pcbnew module such that it always matches the newest KiCAD version (at the moment, v7). However, some of the monkey patching is not backward-compatible.

dunk-law, 2023-01-27:
Report as follows:-
[image]

yaqwsx, 2023-01-27:
And what version of pcbnewTransition is installed on your system?

dunk-law, 2023-01-27:
Don't know that I have it installed. Please clarify?

yaqwsx, 2023-02-03:
What version of IBOM do you use? I just tried installing everything from PCM and it works on my system.

yaqwsx, 2023-02-20:
096e783 should solve the issue, and I was no longer able to reproduce the problem.

closed 2023-02-20 by yaqwsx