yaqwsx/KiKit #487: Error generated when using 'fab jlcpcb --assembly' with KiCad v7

jj-uk, 2023-01-29:
### Prerequisites

- [X] I have read [FAQ](https://github.com/yaqwsx/KiKit/blob/master/doc/faq.md)
- [X] I use the KiKit at least 1.1.0 (older version are not supported)

### KiKit version

kikit, version 1.2.0+24.gdc3206b

### KiCAD version

Application: KiCad x64 on x64  Version: (7.0.0-rc2-26-ga0236113da), release build  Libraries: 	wxWidgets 3.2.1 	FreeType 2.12.1 	HarfBuzz 5.0.1 	FontConfig 2.14.1 	libcurl/7.83.1-DEV Schannel zlib/1.2.13  Platform: Windows 10 (build 19045), 64-bit edition, 64 bit, Little endian, wxMSW  Build Info: 	Date: Jan 27 2023 06:55:05 	wxWidgets: 3.2.1 (wchar_t,wx containers) 	Boost: 1.80.0 	OCC: 7.6.2 	Curl: 7.83.1-DEV 	ngspice: 38 	Compiler: Visual C++ 1934 without C++ ABI  Build settings: 	KICAD_SPICE=ON

### Operating system

Win10

### Description

With KiCad 7, an error is generated when using the --assembly option.

Gerbers are generated correctly.

Error:

C:\Users\joeba\OneDrive\Documents\KiCad\7.0>kikit fab jlcpcb --debug --assembly --no-drc --schematic "C:\dev\v1.0\relay_3ch_spotlights.kicad_sch" "C:\dev\v1.0\panel\panel.kicad_pcb" "C:\dev\v1.0\panel\plots"
An error occurred: can only concatenate str (not "NoneType") to str
No output files produced
Traceback (most recent call last):
  File "C:\Program Files\KiCad\7.0\bin\Lib\runpy.py", line 197, in _run_module_as_main
    return _run_code(code, main_globals, None,
  File "C:\Program Files\KiCad\7.0\bin\Lib\runpy.py", line 87, in _run_code
    exec(code, run_globals)
  File "C:\Users\joeba\OneDrive\Documents\KiCad\7.0\3rdparty\Python39\Scripts\kikit.exe\__main__.py", line 7, in <module>
  File "C:\Program Files\KiCad\7.0\bin\Lib\site-packages\click\core.py", line 1130, in __call__
    return self.main(*args, **kwargs)
  File "C:\Program Files\KiCad\7.0\bin\Lib\site-packages\click\core.py", line 1055, in main
    rv = self.invoke(ctx)
  File "C:\Program Files\KiCad\7.0\bin\Lib\site-packages\click\core.py", line 1657, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
  File "C:\Program Files\KiCad\7.0\bin\Lib\site-packages\click\core.py", line 1657, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
  File "C:\Program Files\KiCad\7.0\bin\Lib\site-packages\click\core.py", line 1404, in invoke
    return ctx.invoke(self.callback, **ctx.params)
  File "C:\Program Files\KiCad\7.0\bin\Lib\site-packages\click\core.py", line 760, in invoke
    return __callback(*args, **kwargs)
  File "C:\Users\joeba\OneDrive\Documents\KiCad\7.0\3rdparty\Python39\site-packages\kikit\fab_ui.py", line 51, in jlcpcb
    return execute(jlcpcb.exportJlcpcb, kwargs)
  File "C:\Users\joeba\OneDrive\Documents\KiCad\7.0\3rdparty\Python39\site-packages\kikit\fab_ui.py", line 30, in execute
    raise e from None
  File "C:\Users\joeba\OneDrive\Documents\KiCad\7.0\3rdparty\Python39\site-packages\kikit\fab_ui.py", line 24, in execute
    return fab(**kwargs)
  File "C:\Users\joeba\OneDrive\Documents\KiCad\7.0\3rdparty\Python39\site-packages\kikit\fab\jlcpcb.py", line 85, in exportJlcpcb
    components = extractComponents(schematic)
  File "C:\Users\joeba\OneDrive\Documents\KiCad\7.0\3rdparty\Python39\site-packages\kikit\fab\common.py", line 22, in extractComponents
    return eeschema_v6.extractComponents(filename)
  File "C:\Users\joeba\OneDrive\Documents\KiCad\7.0\3rdparty\Python39\site-packages\kikit\eeschema_v6.py", line 151, in extractComponents
    symbols, instances = collectSymbols(filename)
  File "C:\Users\joeba\OneDrive\Documents\KiCad\7.0\3rdparty\Python39\site-packages\kikit\eeschema_v6.py", line 124, in collectSymbols
    f = dirname + "/" + f
TypeError: can only concatenate str (not "NoneType") to str

### Steps to Reproduce

Run command on v7 console:

kikit fab jlcpcb --debug --assembly --no-drc --schematic "C:\dev\v1.0\relay_3ch_spotlights.kicad_sch" "C:\dev\v1.0\panel\panel.kicad_pcb" "C:\dev\v1.0\panel\plots"

yaqwsx, 2023-02-03:
Just to create the cross-reference to relevant discussion: #481

yaqwsx, 2023-02-20:
This should be resolved by 32a26eef2dca797f7659a6734085f37eff0902be

closed 2023-02-20 by yaqwsx

jj-uk, 2023-02-20:
Confirmed working for me in a multi-sheet project. Thanks!