yaqwsx/KiKit #488: KiKit action button not visible

murwood, 2023-01-30:
### Prerequisites

- [X] I have read [FAQ](https://github.com/yaqwsx/KiKit/blob/master/doc/faq.md)
- [X] I use the KiKit at least 1.1.0 (older version are not supported)

### KiKit version

version 1.2.0

### KiCAD version

Application: KiCad PCB Editor (64-bit)  Version: (6.0.4), release build  Libraries: 	wxWidgets 3.1.5 	libcurl/7.78.0-DEV Schannel zlib/1.2.11  Platform: Windows 8.1 (build 9600), 64-bit edition, 64 bit, Little endian, wxMSW  Build Info: 	Date: Mar 18 2022 03:04:44 	wxWidgets: 3.1.5 (wchar_t,wx containers) 	Boost: 1.76.0 	OCC: 7.6.0 	Curl: 7.78.0-DEV 	ngspice: 36 	Compiler: Visual C++ 1928 without C++ ABI  Build settings: 	KICAD_USE_OCC=ON 	KICAD_SPICE=ON

### Operating system

Windows 10

### Description

After installation the KiKit action button is not visible, only the KiKit: Show/hide references button. As shown here:
[image]

The only way I found to see it is to go into Preferences > Action Plugins and unselect Show Button for KiKit: Show/hide references.

Then you get:
[image]

### Steps to Reproduce

Install KiKit and open the PCB Editor.

murwood, 2023-01-30:
> Is this your first installation of KiKit? By default, the icon should be visible. However, if you have ever installed KiKit before, it is possible that KiCAD remembers to hide it (it used to be the default).
>
> At the moment, I am unable to reproduce the behavior you describe.
>
> https://github.com/yaqwsx/KiKit/issues/488#issuecomment-1409064669

No, updated yesterday.

Check your PCB Editor>Preferences>Common>Icon Scale. 
That is the problem. If Automatic it cuts off the Kikit buttons!

So the problem is in PCB Editor not KiKit.

Can you verify this?

Tom

yaqwsx, 2023-02-26:
I wasn't able to reproduce this behavior. I assume, this a KiCAD problem, not KiKit's.

closed 2023-02-26 by yaqwsx