yaqwsx/KiKit #489: KiKit GUI too tall, Panelize button hidden

murwood, 2023-01-30:
### Prerequisites

- [X] I have read [FAQ](https://github.com/yaqwsx/KiKit/blob/master/doc/faq.md)
- [X] I use the KiKit at least 1.1.0 (older version are not supported)

### KiKit version

version 1.2.0

### KiCAD version

Application: KiCad PCB Editor (64-bit)  Version: (6.0.4), release build  Libraries: 	wxWidgets 3.1.5 	libcurl/7.78.0-DEV Schannel zlib/1.2.11  Platform: Windows 8.1 (build 9600), 64-bit edition, 64 bit, Little endian, wxMSW  Build Info: 	Date: Mar 18 2022 03:04:44 	wxWidgets: 3.1.5 (wchar_t,wx containers) 	Boost: 1.76.0 	OCC: 7.6.0 	Curl: 7.78.0-DEV 	ngspice: 36 	Compiler: Visual C++ 1928 without C++ ABI  Build settings: 	KICAD_USE_OCC=ON 	KICAD_SPICE=ON

### Operating system

Windows 10

### Description

When you press the KiKit action button and the KiKit GUI opens, the Panelize button is hidden by the task bar and you can not get to it, as shown here:
[image]

If you hide the taskbar, it will be available. But only the first time. Close the GUI and e-open and it resizes larger and the Panelize button is hidden again, as shown here:
[image]

### Steps to Reproduce

Open PCB Editor and press the KiKit action button.

yaqwsx, 2023-02-01:
Could you open Scipting console from the PCB editor, enter the following commands and give me the output as shown below?
```
import wx
wx.Display().GetClientArea()
```

[image]

murwood, 2023-02-01:
Here is what I get:
wx.Rect(0, 0, 1600, 860)

Tom

https://github.com/yaqwsx/KiKit/issues/489#issuecomment-1412647358

CapnKernel, 2024-01-10:
If anyone is having this problem under Linux, then in most Linux window managers, you can hold down Alt, then click and drag the window until the panelise button is visible.