yaqwsx/KiKit #491: assert len(symbols) == len(instances) error

greenscreenflicker, 2023-02-03:
When exporting the following pcb 
[pcb.zip](https://github.com/yaqwsx/KiKit/files/10579112/pcb.zip)

using the command
>kikit fab jlcpcb --no-drc --assembly --schematic Flyback_Converter.kicad_sch Flyback_Converter.kicad_pcb gerber/

i get the following error

[image]

greenscreenflicker, 2023-02-03:
Notify @yaqwsx

yaqwsx, 2023-02-03:
Please, don't deliberately ignore issue templates. They are here for a reason - you would have found out that you just use old KiKit version.

The bug you observe is not present on the current release or upstream version. This bug was fixed in https://github.com/yaqwsx/KiKit/commit/933cbf9027fda067fb094539856e414cf06e912a.

closed 2023-02-03 by yaqwsx