yaqwsx/KiKit #497: KiKit crashes under V7: AttributeError: module 'pcbnew' has no attribute 'DRAWSEGMENT'

greenscreenflicker, 2023-02-13:
### Prerequisites

- [X] I have read [FAQ](https://github.com/yaqwsx/KiKit/blob/master/doc/faq.md)
- [X] I use the KiKit at least 1.2.0 (older version are not supported)

### KiKit version

no output

### KiCAD version

7.0.0

### Operating system

Windows 

### Description

Executing the script under kicad v7 returns the following error:

`C:\Users\mchhe\OneDrive\Dokumente\git\dpsbrake\Brake_Power_Supply\v5>kikit --version
Traceback (most recent call last):
  File "C:\Program Files\KiCad\7.0\bin\Lib\runpy.py", line 197, in _run_module_as_main
    return _run_code(code, main_globals, None,
  File "C:\Program Files\KiCad\7.0\bin\Lib\runpy.py", line 87, in _run_code
    exec(code, run_globals)
  File "C:\Users\mchhe\OneDrive\Dokumente\KiCad\7.0\3rdparty\Python39\Scripts\kikit.exe\__main__.py", line 4, in <module>
  File "C:\Users\mchhe\OneDrive\Dokumente\KiCad\7.0\3rdparty\Python39\site-packages\kikit\ui.py", line 2, in <module>
    from kikit import (panelize_ui, export_ui, present_ui, stencil_ui,
  File "C:\Users\mchhe\OneDrive\Dokumente\KiCad\7.0\3rdparty\Python39\site-packages\kikit\panelize_ui.py", line 7, in <module>
    from kikit.panelize_ui_sections import *
  File "C:\Users\mchhe\OneDrive\Dokumente\KiCad\7.0\3rdparty\Python39\site-packages\kikit\panelize_ui_sections.py", line 4, in <module>
    from kikit import plugin
  File "C:\Users\mchhe\OneDrive\Dokumente\KiCad\7.0\3rdparty\Python39\site-packages\kikit\plugin.py", line 10, in <module>
    from kikit.panelize import Panel
  File "C:\Users\mchhe\OneDrive\Dokumente\KiCad\7.0\3rdparty\Python39\site-packages\kikit\panelize.py", line 3, in <module>
    from pcbnewTransition import pcbnew, isV6
  File "C:\Users\mchhe\OneDrive\Dokumente\KiCad\7.0\3rdparty\Python39\site-packages\pcbnewTransition\__init__.py", line 6, in <module>
    from .transition import *
  File "C:\Users\mchhe\OneDrive\Dokumente\KiCad\7.0\3rdparty\Python39\site-packages\pcbnewTransition\transition.py", line 50, in <module>
    pcbnew.PCB_SHAPE = pcbnew.DRAWSEGMENT
AttributeError: module 'pcbnew' has no attribute 'DRAWSEGMENT'

C:\Users\mchhe\OneDrive\Dokumente\git\dpsbrake\Brake_Power_Supply\v5>`

### Steps to Reproduce

Execute kikit --version

greenscreenflicker, 2023-02-13:
Notify @yaqwsx

greenscreenflicker, 2023-02-13:
sorry for the dump question: how to get the upsteam release?

yaqwsx, 2023-02-13:
https://github.com/yaqwsx/KiKit/blob/master/doc/installation.md#installing-a-special-version-of-kikit