yaqwsx/KiKit #498: on KiCAD 7.0.0 it doesn't work

tormyvancool, 2023-02-13:
### Prerequisites

- [X] I have read [FAQ](https://github.com/yaqwsx/KiKit/blob/master/doc/faq.md)
- [X] I use the KiKit at least 1.2.0 (older version are not supported)

### KiKit version

1.2.0

### KiCAD version

7.0.0

### Operating system

Windows 11 Pro 64 bits

### Description

Firsto of all:
I have the 1.1.0 because teh 1.2.0 doesn't even appear available on the Plugin Manager of KiCAD
So, if this but there is not in the 1.2.0, please make it available

This window still appear despite the procedure described here in GitHub
[image]

### Steps to Reproduce

just start KiCAD
Start PCBNew or FootPrint Editor.

closed 2023-02-13 by yaqwsx

tormyvancool, 2023-02-13:
> This is duplicate of #497.

NO it doesn't crash ... it doesn't work at all

yaqwsx, 2023-02-14:
> The current release doesn't support v7. The bug is not present on the upstream version, but there wasn't a release for v7, as we have one more bug to fix (#487). It should be out in a week or so. But in the meantime, you can install the upstream version and use KiKit with V7.