yaqwsx/KiKit #499: Fontconfig / Object error

greenscreenflicker, 2023-02-15:
### Prerequisites

- [X] I have read [FAQ](https://github.com/yaqwsx/KiKit/blob/master/doc/faq.md)
- [X] I use the KiKit at least 1.2.0 (older version are not supported)

### KiKit version

kikit, version 1.2.0+42.g1135c40

### KiCAD version

7.0.0

### Operating system

Windows

### Description

When trying to produce kicad output files i get the error:

kikit fab jlcpcb --no-drc --assembly --schematic Brake_Power_Supply.kicad_sch Brake_Power_Supply.kicad_pcb gerber/
Fontconfig error: Cannot load default config file: No such file: (null)
Fontconfig error: Cannot load default config file: No such file: (null)
An error occurred: expected str, bytes or os.PathLike object, not NoneType
No output files produced

### Steps to Reproduce

Execute commend above

yaqwsx, 2023-02-20:
There are two problems:
- regarding font. I believe this must be misconfigured fonts in your system (and the errors are present anytime you load KiCAD), and the problem has nothing to do with KiKit.
- `An error occurred: expected str, bytes or os.PathLike object, not NoneType`: you see a duplicate of #487

closed 2023-02-20 by yaqwsx

greenscreenflicker, 2023-02-21:
@yaqwsx Works now. Thanks.