yaqwsx/KiKit #500: Kikit doesn't work with kicad v7

troisiemetype, 2023-02-15:
### Prerequisites

- [X] I have read [FAQ](https://github.com/yaqwsx/KiKit/blob/master/doc/faq.md)
- [X] I use the KiKit at least 1.2.0 (older version are not supported)

### KiKit version

I cannot check it !

### KiCAD version

7.0.0

### Operating system

Lubuntu

### Description

Hello,

Short : kikit doesn't work with kicad v7 on my PC. I've read the various issues reported about v7, but I can find a solution with the.

Trying to use kikit, I have this error : 
`ModuleNotFoundError: No module named 'pcbnew'`
I've read quite a few things about this error, which seems have happened with v5 and v6 as well.
I've tried to add kicad python library's path to python, it finds the library, but then I have :
`ImportError: libkicad_3dsg.so.2.0.0: cannot open shared object file: No such file or directory`

I've read that python should be executed from the kicad CLI, but I didn't understand how to do it.
I've tried to install the upstream release as stated in #497, but pip refuses to install it because :
`      No Pcbnew Python module found for the current Python interpreter.`

I was using kikit, installed from pip, with v6 from the command line. I'm a bit lost at what I can do here.

Note : I cannot check kikit version because it refuses to run even for that. It displays the same error message.

Here is the complete error traceback : 
```
Traceback (most recent call last):
  File "/home/pierreloupm/.local/bin/kikit", line 5, in <module>
    from kikit.ui import cli
  File "/home/pierreloupm/.local/lib/python3.10/site-packages/kikit/ui.py", line 2, in <module>
    from kikit import (panelize_ui, export_ui, present_ui, stencil_ui,
  File "/home/pierreloupm/.local/lib/python3.10/site-packages/kikit/panelize_ui.py", line 7, in <module>
    from kikit.panelize_ui_sections import *
  File "/home/pierreloupm/.local/lib/python3.10/site-packages/kikit/panelize_ui_sections.py", line 4, in <module>
    from kikit import plugin
  File "/home/pierreloupm/.local/lib/python3.10/site-packages/kikit/plugin.py", line 10, in <module>
    from kikit.panelize import Panel
  File "/home/pierreloupm/.local/lib/python3.10/site-packages/kikit/panelize.py", line 3, in <module>
    from pcbnewTransition import pcbnew, isV6
  File "/home/pierreloupm/.local/lib/python3.10/site-packages/pcbnewTransition/__init__.py", line 6, in <module>
    from .transition import *
  File "/home/pierreloupm/.local/lib/python3.10/site-packages/pcbnewTransition/transition.py", line 2, in <module>
    import pcbnew
ModuleNotFoundError: No module named 'pcbnew'
```
I'm sorry if it is more relevant for kicad than kikit.

Thank you.

### Steps to Reproduce

Install kicad v7 ?

yaqwsx, 2023-02-15:
There's a problem with KiCAD 7 packaging on Ubuntu: https://gitlab.com/kicad/packaging/kicad-ubuntu-builder/kicad-daily-package/-/issues/43

troisiemetype, 2023-02-15:
I'm really grateful for this fast answer ! And sorry for adding to the list of issues for a problem that is elsewhere.
At least this one can be closed. I hope that on the kicad side they will soon find the problem and push an update.

wero1414, 2023-02-17:
i have the same problem on MacOS with Intel

troisiemetype, 2023-02-18:
I don't know if the issue mentioned above is in the right place, nobody has answered it yet. Maybe another one should be opened by declaring a bug from inside kicad ?

mikhailmihalkov, 2023-02-18:
Win 10 plugin is installed but kicad 7 does not see it

yaqwsx, 2023-02-19:
> Win 10 plugin is installed but kicad 7 does not see it

What KiKit version do you use? Only the upstream version works with v7.

yaqwsx, 2023-02-19:
> I don't know if the issue mentioned above is in the right place, nobody has answered it yet. Maybe another one should be opened by declaring a bug from inside kicad ?

Feel free to open issue on other places. I won't be able to open a proper, fully descriptive issue in the upcoming days.

mikhailmihalkov, 2023-02-19:
> > Я не знаю, находится ли вопрос, упомянутый выше, в нужном месте, никто еще не ответил на него. Может быть, следует открыть еще один, объявив об ошибке изнутри kicad?
> 
> Не стесняйтесь открывать вопрос в других местах. Я не смогу открыть надлежащий, полностью описательный выпуск в ближайшие дни.
I do everything according to the instructions as for version 6 in version 7 pip install kikit Do you need to do something else?

yaqwsx, 2023-02-19:
@mikhailmihalkov: "**Only the upstream version works with v7.**" - not the released one. Read the installation instructions: https://github.com/yaqwsx/KiKit/blob/master/doc/installation.md#installing-a-special-version-of-kikit

mikhailmihalkov, 2023-02-19:
Thank you for trying.. Do I understand the old one from the plugin manager needs to be removed?

kongr45gpen, 2023-02-20:
Commands that helped my Kikit recognise the current KiCAD distribution (taken from https://github.com/yaqwsx/KiKit/commit/38a751977c0a749697c63c50083c13746938856c):

```bash
export PYTHONPATH=/usr/lib/kicad/lib/python3/dist-packages:${PYTHONPATH}
export LD_LIBRARY_PATH=/usr/lib/kicad/lib/x86_64-linux-gnu/:${LD_LIBRARY_PATH}
python3 -m pip install git+https://github.com/yaqwsx/KiKit@master
```

troisiemetype, 2023-02-20:
OH YEAH !!
It works for me too. Thank you !

yaqwsx, 2023-02-20:
As @kongr45gpen mentioned, I dug into the problem, and it seems that the KiCAD developers decided to move the python module into non-standard locations. The upstream version should now be fully compatible with KiCAD 7.

The installation and KiKit itself should be fixed in 2d859c5d8fceb1cf472ac4500398a745a48517ed and 9390b70c784c735de6d9c53a2fd9aabe927a5184

closed 2023-02-20 by yaqwsx

mikhailmihalkov, 2023-02-20:
> Как@kongr45gpenРазработчики KiCAD решили переместить модуль python в нестандартные места. Исходная версия теперь должна быть полностью совместима с KiCAD 7.
> 
> Установка и сам KiKit должен быть исправлен в [2d859c5](https://github.com/yaqwsx/KiKit/commit/2d859c5d8fceb1cf472ac4500398a745a48517ed) и [9390b70](https://github.com/yaqwsx/KiKit/commit/9390b70c784c735de6d9c53a2fd9aabe927a5184)
What about win 10?

yaqwsx, 2023-02-20:
There was never such a problem with Windows 10.

mikhailmihalkov, 2023-02-20:
> Никогда не было таких проблем с Windows 10.
Do I fail to install in win 10 a special version of kikit you have plans to add a fixed version to the plugin manager?

yaqwsx, 2023-02-20:
Yes, of course there will be a release (why wouldn't there be one when we put so much effort in supporting v7?). But it needs proper testing before it gets released. This takes time and effort.

matiasbarone, 2024-01-11:
I've kicad v7.0.2 installed with flatpak.

If I install it in my sistem gives me the error: 
`ModuleNotFoundError: No module named 'pcbnew'`

If I install kikit inside the flatpak I have this error:
```
[📦 org.kicad.KiCad ~]$ python3 -m kikit.ui --help
/usr/bin/python3: Error while finding module specification for 'kikit.ui' (ModuleNotFoundError: No module named 'kikit')
```

yaqwsx, 2024-01-12:
Flatpak is sandboxed. So whatever you install out of it, flatpak KiCAD and Python cannot see. In Flatpak, there is no python3 - there is only python. The python3 probably refers to your system's python.

In flatpak, you should do:
```
python -mpip install kikit
python -mkikit.ui --help
```