yaqwsx/KiKit #503: Assertion error on panelize.appendBoard()

troisiemetype, 2023-02-21:
### Prerequisites

- [X] I have read [FAQ](https://github.com/yaqwsx/KiKit/blob/master/doc/faq.md)
- [X] I use the KiKit at least 1.2.0 (older version are not supported)
- [X] I understand that version 1.2.0 is not compatible with KiCAD 7 and if this is a bug report about v7 incompatibility, I tried the latest upstream version first

### KiKit version

kikit, version 1.2.0+52.g125c752

### KiCAD version

7.0.0

### Operating system

Lubuntu 22.04 LTS

### Description

I have an error while running a script that worked ok with the v6 : 
```

Traceback (most recent call last):
  File "/home/pierreloupm/Documents/kicad/pod/panels/./panelize.py", line 146, in <module>
    main()
  File "/home/pierreloupm/Documents/kicad/pod/panels/./panelize.py", line 53, in main
    bounding_box_1 = panel.appendBoard(
  File "/home/pierreloupm/.local/lib/python3.10/site-packages/kikit/panelize.py", line 881, in appendBoard
    drawings = collectItems(board.GetDrawings(), enlargedSourceArea)
  File "/home/pierreloupm/.local/lib/python3.10/site-packages/kikit/common.py", line 82, in collectItems
    return list([x for x in boardCollection if fitsIn(x.GetBoundingBox(), sourceArea)])
  File "/home/pierreloupm/.local/lib/python3.10/site-packages/kikit/common.py", line 82, in <listcomp>
    return list([x for x in boardCollection if fitsIn(x.GetBoundingBox(), sourceArea)])
  File "/home/pierreloupm/.local/lib/python3.10/site-packages/kikit/common.py", line 46, in fitsIn
    assert isinstance(what, (BOX2I, VECTOR2I, pcbnew.wxPoint))
AssertionError
```

Here is the script that causes this error.
[panelize.zip](https://github.com/yaqwsx/KiKit/files/10788434/panelize.zip)
And here are the two boards I need to merge into a panel.
[pod.zip](https://github.com/yaqwsx/KiKit/files/10788454/pod.zip)

I've added the kicad library path as specified here : https://github.com/yaqwsx/KiKit/issues/500#issuecomment-1436714529

I hope it will help.
Thank you.

### Steps to Reproduce

Run the script on the two kicad_pcb files.

troisiemetype, 2023-02-22:
It indeed works !
But other problems then arise...
```
Traceback (most recent call last):
  File "/home/pierreloupm/Documents/kicad/pod/panels/./panelize.py", line 148, in <module>
    main()
  File "/home/pierreloupm/Documents/kicad/pod/panels/./panelize.py", line 66, in main
    bounding_box_2 = panel.appendBoard(
  File "/home/pierreloupm/.local/lib/python3.10/site-packages/kikit/panelize.py", line 841, in appendBoard
    raise RuntimeError("Board rotation has to be passed as EDA_ANGLE, not a number")
RuntimeError: Board rotation has to be passed as EDA_ANGLE, not a number
```

That I've solved by changing `rotationAngle = 0` to `rotationAngle = EDA_ANGLE(0,1)` in `panel.appendBoard()`.

Then
```
Traceback (most recent call last):
  File "/home/pierreloupm/Documents/kicad/pod/panels/./panelize.py", line 149, in <module>
    main()
  File "/home/pierreloupm/Documents/kicad/pod/panels/./panelize.py", line 84, in main
    framing_substrates = ki.dummyFramingSubstrate(
  File "/home/pierreloupm/.local/lib/python3.10/site-packages/kikit/panelize_ui_impl.py", line 392, in dummyFramingSubstrate
    framingPreset = preset["framing"]
TypeError: tuple indices must be integers or slices, not str
```
That I unfortunately don't know how to solve.
If I comment it, I have another error from `panel.buildTabsFromAnnotations()` which now wants the millFillets radius as an argument. It's straight forward, no problem here. 

Then I have this last error : 
```
Traceback (most recent call last):
  File "/home/pierreloupm/Documents/kicad/pod/panels/./panelize.py", line 149, in <module>
    main()
  File "/home/pierreloupm/Documents/kicad/pod/panels/./panelize.py", line 145, in main
    panel.save()
  File "/home/pierreloupm/.local/lib/python3.10/site-packages/kikit/panelize.py", line 519, in save
    self.transferProjectSettings()
  File "/home/pierreloupm/.local/lib/python3.10/site-packages/kikit/panelize.py", line 616, in transferProjectSettings
    raise RuntimeError("Merging of DRC rules of multiple boards is currently unsupported")
RuntimeError: Merging of DRC rules of multiple boards is currently unsupported
```
Which I don't know if it's something that has not yet being adapted to kicad7, or if it was already the case. I believe this bit worked with v6. Anyway, in my case I can also import my boards from a single file, as that's where they are in the first place,and then it works.

It seems each new version of kicad brings its load of work !

Note : this morning a new release from kicad was available for installation through Apt, so it's now v7.0.0-1. I was sadly not able to find a log of what has been fixed in this release.

yaqwsx, 2023-02-22:
I am sorry for the trouble; as I haven't prepared proper scripting examples and #283 isn't finished yet, I didn't care much about scripting compatibility:
- With KiCAD 7 support, we migrated all angle values to `EDA_ANGLE`, when you want to create an angle in degrees, you should use `pcbnew.EDA_ANGLE(42, pcbnew.DEGREES_T)`
- based on `TypeError: tuple indices must be integers or slices, not str` the problem is that you pass tuple of width and height, instead of a preset (dictionary). All methods in the `panelize_ui_impl` module work with CLI presets. As a part of #283, we will factor some of the functionality out and make it more scripting friendly.
- at the moment, we do not support merging multiple DRC rules merging (#36). Net classes are transferred, but you have to choose one of the board to be the source of global DRC rules and ignore it for the rest (set argument `inheritDrc` of `appendBoard` to false).

troisiemetype, 2023-02-22:
I'm sorry to open that much issues too !

Thank you for the tip for EDA_ANGLE : I tried to pass `EDA_ANGLE.DEGREES_T` and `EDA_ANGLE_T.DEGREES_T`,without success. I finally passed `1` instead, which work. I must admit I don't know Python enough to know exactly how to do things. But in the end it works. :)

Ok for DRC rules. I was precisely looking at the `appendBoard()` function, chasing another problem, seen it and thought it should solve the problem. Which it does, indeed.

For the preset, ok. I understand that I should construct a `preset` object, which is a dictionary composed of all the options available, each of this option being a sub-dictionary. For the purpose of the `dummyFramingSubstrate()`, I understand that having just the `"framing"` entry, as described in `panelize_ui_sections.py`, would do it ?
That would be something like this ?
```
preset = {
    "framing" : 
        "type": "frame",
        "space": my_value,
}
```
I will give it a try this evening.

PS : I'm not in a hurry : the boards I needed it for have been sent to fab as two independent boards,and you may have more important issues to work with than this one. :)

BillyPocketNC, 2023-06-28:
Modifying the value of the board annotation footprint will cause this error as well.
[image]
changing the value back to Board fixed the error.

Thanks for building kikit @yaqwsx it saves me a lot of time when I don't do dumb things.

> I have an error while running a script that worked ok with the v6 :
>
> ```
>
> Traceback (most recent call last):
>   File "/home/pierreloupm/Documents/kicad/pod/panels/./panelize.py", line 146, in [...]