yaqwsx/KiKit #504: Assembly Options

greenscreenflicker, 2023-02-21:
### Prerequisites

- [X] I have read the documentation and the proposed feature is not implemented.

### Description

Sometimes I have the requirement to have different assembly options.
So it would be great, to panalize PCB with several PCBs. Is there currently any possibility?

yaqwsx, 2023-02-21:
I am not sure if I understand - the feature request mixes two feature:
- multi-design panels (at the moment available via scripting (#283), there are plans for scripting-less multi design panels, see (#327).
- various assembly options are supported via multiple order codes. See documentation for the fab command.

Or do I miss the actual feature request?

yaqwsx, 2023-06-27:
Closing for inactivity

closed 2023-06-27 by yaqwsx