yaqwsx/KiKit #505: KiKit V7: Do not place option is not respected.

greenscreenflicker, 2023-02-21:
### Prerequisites

- [X] I have read [FAQ](https://github.com/yaqwsx/KiKit/blob/master/doc/faq.md)
- [X] I use the KiKit at least 1.2.0 (older version are not supported)
- [X] I understand that version 1.2.0 is not compatible with KiCAD 7 and if this is a bug report about v7 incompatibility, I tried the latest upstream version first

### KiKit version

kikit, version 1.2.0+52.g125c752

### KiCAD version

7.0.0

### Operating system

Windows

### Description

In Kicad7 components can be marked as Do Not Poplulate / Do not place. 
This option is not respected.

### Steps to Reproduce

1. Mark component in V7 as DNP
2. A warning is still issued when no LCSC partnumber is added when DNP option is on.
3. Probabbly (I didn't test it) it's also in the BOM.

yaqwsx, 2023-02-21:
At the moment, we respect the "exclude from bom" option. But respecting the "exclude from board" for these services also makes sense. Implemented in 042493c.

closed 2023-02-21 by yaqwsx

greenscreenflicker, 2023-02-24:
@yaqwsx Now i found time to test.

Version: kikit, version 1.2.0+67.gbb5ad7f

Apparently, I was not clear enough. I mean the do not popuplate option.
[image]
And it still gives me the following error msg:
`WARNING: Component TP703 is missing ordercode`

yaqwsx, 2023-02-24:
Try a9ed39d