yaqwsx/KiKit #506: Upstream Version - tab radius not accepting any values other than zero

bandgapltd, 2023-02-21:
### Prerequisites

- [X] I have read [FAQ](https://github.com/yaqwsx/KiKit/blob/master/doc/faq.md)
- [X] I use the KiKit at least 1.2.0 (older version are not supported)
- [X] I understand that version 1.2.0 is not compatible with KiCAD 7 and if this is a bug report about v7 incompatibility, I tried the latest upstream version first

### KiKit version

kikit, version 1.2.0+52.g125c752

### KiCAD version

V7

### Operating system

Windows 10

### Description

The plugin is working except for the tab radius setting. I don't know whether this has been implemented or is expected to work, but any value I've tried throws up an error.

The board I'm trying to panelise has edge.cuts in a footprint, but seems to work for all other extensive purposes.

### Steps to Reproduce

Set tab radius to anything other than default of 0mm.
[image]

yaqwsx, 2023-02-21:
The feature is supposed to work. However, to properly debug the issue, I have to be able to reproduce it. Could you share the board file where it fails (outlines are sufficient).

bandgapltd, 2023-02-21:
Hi Jan. Okay will send over shortly. Thanks

bandgapltd, 2023-02-21:
[TVSBoardOutline1_0.kicad_mod.txt](https://github.com/yaqwsx/KiKit/files/10797754/TVSBoardOutline1_0.kicad_mod.txt)
[outline.kicad_pcb.txt](https://github.com/yaqwsx/KiKit/files/10797757/outline.kicad_pcb.txt)

bandgapltd, 2023-02-21:
have appended .txt to the footprint and board files to allow uploading. The outline for the board is in this kicad footprint as it was created from a DXF.

bandgapltd, 2023-02-21:
[image]

As you can see, without the tab radius it is working properly.

yaqwsx, 2023-02-21:
The problem is a numerical instability of Shapely. There are plans for resolving this, but it won't be anytime soon. 

In the meantime, I suggest to use millradius in the postprocess section that will serve you well (and faster).

yaqwsx, 2023-02-21:
...actually, 266a8ae, might be a doable "hack" to overcome this instability.

closed 2023-02-21 by yaqwsx

bandgapltd, 2023-02-21:
Will take a look at the link you included. Millradius worked perfectly though and so I'm happy with that. Time to get my GERBERs sent to the fab house. Thanks for your help.