yaqwsx/KiKit #511: Latest docker nightly not working with KiCAD 7 release any longer :(

UnexpectedMaker, 2023-02-26:
### Prerequisites

- [X] I have read [FAQ](https://yaqwsx.github.io/KiKit/latest/faq/)
- [X] I use the KiKit at least 1.2.0 (older version are not supported)
- [X] I understand that version 1.2.0 is not compatible with KiCAD 7 and if this is a bug report about v7 incompatibility, I tried the latest upstream version first

### KiKit version

Latest docker nightly (21 hours ago)

### KiCAD version

Version: (7.0.0-0), release build

### Operating system

macOS 13.2.1

### Description

Trying to panellise a board with the latest nightly docker image, and I am getting the following error:
```
An error occurred: KiCad was unable to open this file because it was created with a more recent version than the one you are running.

To open it you will need to upgrade KiCad to a version dated 10/18/22 or later.

Full error text:
KiCad was unable to open this file because it was created with a more recent version than the one you are running.

To open it you will need to upgrade KiCad to a version dated 10/18/22 or later.

Full error text:
Expecting 'layer, width, fill, tstamp, locked, or status' in '/KICAD/Neo7/Neo7 Original/NeoSevenSegment.kicad_pcb', line 125, offset 8.
No output files produced
```

But this was made in 7.0 release version - and kikit nightlies were working in the 7.0 RC builds prior to this.

### Steps to Reproduce

Download the latest kikit nightly docker image, run it and try to panelist a board made in KiCAD 7.0 release.

yaqwsx, 2023-02-26:
Try the new `-v7` images that are built using KiCAD 7 (the current nightly was built using v6): https://hub.docker.com/repository/docker/yaqwsx/kikit/general

yaqwsx, 2023-02-27:
I assume this is fixed now.