yaqwsx/KiKit #512: Mac OSX install script broken in KiCad 7

tanaes, 2023-02-28:
### Prerequisites

- [X] I have read [FAQ](https://yaqwsx.github.io/KiKit/latest/faq/)
- [X] I use KiKit at least version 1.3.0 (older version are not supported)

### KiKit version

N/A

### KiCAD version

7.0.0

### Operating system

Mac OSX

### Description

Recently updated to KiCad 7.0, which broke my KiKit install. Decided to go ahead and just reinstall following [these directions](https://yaqwsx.github.io/KiKit/v1.3/installation/macos/). 

This throws an error:

```
Jons-MacBook-Pro:~ jgs286$ sudo bash installMacOS.bash 
Kicad Python dir: /Applications/KiCad/KiCad.app/Contents/Frameworks/Python.framework/Versions/3.8
Python version: 3.8
Signinf certificate: kikit
KiKit will be installed
installMacOS.bash: line 18: /Applications/KiCad/KiCad.app/Contents/Frameworks/Python.framework/Versions/3.8/bin/python3: No such file or directory
```

Which seems to derive from the specification of Python v3.8, while KiCad 7.0 appears to have v3.9:

install script:

```
V=3.8
KICAD_PYTHON="/Applications/KiCad/KiCad.app/Contents/Frameworks/Python.framework/Versions/$V"
CERTIFICATE="kikit"
```

KiCAD dir: 

```
Jons-MacBook-Pro:~ jgs286$ ls /Applications/KiCad/KiCad.app/Contents/Frameworks/Python.framework/Versions/
3.9/     Current@
```

Replacing `V=3.8` with `V=3.9` in the install script resolved the issue for me. KiCAD 7.0 seems to use the `Current` symlink, so if KiCAD 6.0 also does, perhaps that is an elegant way to get around the error?

### Steps to Reproduce

- Update to KiCad 7.0
- Run installMacOS.bash

Anderson69s, 2023-03-12:
May I add some problem with this configuration? PCBNew is crashing every time I try to use Kikit on OSX. No Problem on windows.
KiCAD V 7.0.1
Kikit V1.3.0
Pc : Win 10 with latest update
MAC : M2 OSX 13.2.1
I can post log files if you need it ;-)

yaqwsx, 2023-03-12:
@tanaes: This should be addressed in #516 

@Anderson69s: Please, collect as many logs as possible. I have no idea what might be causing problems at the moment on Mac

Anderson69s, 2023-03-12:
@yaqwsx Actually, this is wird, I have to admit it... When I make a panel with windows no problems at all.

On mac, if I make a panel in a empty sheet of PCBNew, it work. But once the panel is saved you can't regenerate it. Here is a crash bug from OSX : 
`-------------------------------------
Translated Report (Full Report Below)
-------------------------------------

Process:               pcbnew [1953]
Path:                  /Applications/KiCad/KiCad.app/Contents/Applications/pcbnew.app/Contents/MacOS/pcbnew
Identifier:            org.kicad.pcbnew
Version:               7.0.1-0 (7.0.1-0)
Code Type:             ARM-64 (Native)
Parent Process:        kicad [1890]
Responsible:           kicad [1890]
User ID:               501

Date/Time:             2023-03-12 19:40:59.6528 +0100
OS Version:            macOS 13.2.1 (22D68)
Report Version:        12
Anonymous UUID:        05725341-A445-2D5E-DD4E-3853B27F3463

Sleep/Wake UUID:       DCCD23A4-3789-492B-A3CA-102BA3FC02D8

Time Awake Since Boot: 4400 seconds
Time Since Wake:       471 seconds

System Integrity Protection: enabled

Crashed Thread:        0  Dispatch queue: com.apple.main-thread

Exception Type:        EXC_BAD_ACCESS (SIGSEGV)
Exception Codes:       KERN_INVALID_ADDRESS at 0x0000000000000070
Exception Codes:       0x0000000000000001, 0x0000000000000070

Termination Reason:    Namespace SIGNAL, Code 11 Segmentation fault: 11
Terminating Process:   exc handler [1953]

VM Region Info: 0x70 is not in any region.  Bytes before following region: 105556203274128
      REGION TYPE                    START - END         [ VSIZE] PRT/MAX SHRMOD  REGION DETAIL
      UNUSED SPACE AT START
--->  
      MALLOC_NANO (reserved)   6000b8000000-6000c0000000 [128.0M] rw-/rwx SM=NUL  ...(unallocated)

Kernel Triage:
VM - pmap_enter retried due to resource shortage
VM - pmap_enter retried due to resource shortage
VM - pmap_enter retried due to resource shortage
VM - pmap_enter retried due to resource shortage
VM - pmap_enter retried due to resource shortage

Thread 0 Crashed::  Dispatch queue: com.apple.main-thread
0   _pcbnew.kiface                	       0x1162cac68 BOARD::IsLayerVisible(PCB_LAYER_ID) const + 56
1   _pcbnew.kiface                	       0x1163f7a70 PCB_DRAW_PANEL_GAL::SyncLayersVisibility(BOARD const*) + 132
2   _pcbnew.kiface                	       0x1163f7a70 PCB_DRAW_PANEL_GAL::SyncLayersVisibility(BOARD const*) + 132
3   _pcbnew.kiface                	       0x116196fe4 PCB_EDIT_FRAME::RebuildAndRefresh() + 1276
4   _pcbnew.kiface                	       0x116053370 _wrap_Refresh(_object*, _object*) + 148
5   Python                        	       0x10172a884 cfunction_vectorcall_NOARGS + 288
6   Python                        	       0x1017be93c call_function + 416
7   Python                        	       0x1017bbfdc _PyEval_EvalFrameDefault + 23108
8   Python                        	       0x1016e8cec function_code_fastcall + 112
9   Python                        	       0x1017be93c call_function + 416
10  Python                        	       0x1017bbfdc _PyEval_EvalFrameDefault + 23108
11  Python                        	       0x1016e8cec function_code_fastcall + 112
12  Python                        	       0x1016ead34 method_vectorcall + 396
13  _core.cpython-39-darwin.so    	       0x124928a98 wxPyCallback::EventThunker(wxEvent&) + 364
14  libwx_osx_cocoau-3.2.0.2.1.dylib	       0x1028fa3d4 wxEvtHandler::SearchDynamicEventTable(wxEvent&) + 360
15  libwx_osx_cocoau-3.2.0.2.1.dylib	       0x1028fa078 wxEvtHandler::ProcessEventLocally(wxEvent&) + 88
16  libwx_osx_cocoau-3.2.0.2.1.dylib	       0x1028f9ea0 wxEvtHandler::ProcessEvent(wxEvent&) + 56
17  _core.cpython-39-darwin.so    	       0x124904c64 sipwxDialog::ProcessEvent(wxEvent&) + 128
18  libwx_osx_cocoau-3.2.0.2.1.dylib	       0x102b0a140 wxWindowBase::TryAfter(wxEvent&) + 252
[... 497 lines cut ...]
},
  "trialInfo" : {
  "rollouts" : [
    {
      "rolloutId" : "60186475825c62000ccf5450",
      "factorPackIds" : {

      },
      "deploymentId" : 240000026
    },
    {
      "rolloutId" : "61675b89201f677a9a4cbd65",
      "factorPackIds" : {
        "HEALTH_FEATURE_AVAILABILITY" : "638e768d75b0720ff1f8c4f3"
      },
      "deploymentId" : 240000051
    }
  ],
  "experiments" : [

  ]
}
}

`