yaqwsx/KiKit #513: RuntimeWarning: invalid value encountered in intersection

greenscreenflicker, 2023-03-01:
### Prerequisites

- [X] I have read [FAQ](https://yaqwsx.github.io/KiKit/latest/faq/)
- [X] I use KiKit at least version 1.3.0 (older version are not supported)

### KiKit version

kikit, version 1.3.0+2.ge554018

### KiCAD version

7.0.0

### Operating system

Windows 11

### Description

``kikit panelize  --layout "grid; rows: 1; cols: 1; space: 2mm"  --tabs "fixed; width: 20mm; vcount: 1; hcount: 1" --cuts "mousebites; drill: 0.5mm; spacing: 1mm; offset: 0.2mm; prolong: 0.5mm" --fiducials "3fid; hoffset: 5mm; voffset: 2.5mm; coppersize: 2mm; opening: 1mm;"  --text "simple; text: Digital Power Systems GmbH, Karlsruhe, Brake Power Supply v5, production run1; anchor: mt; voffset: 2.5mm; hjustify: center; vjustify: center;"  --framing "railstb; width: 5mm; space: 1mm;"  --post "millradius: 0.5mm"   --source "tolerance: 20mm"  Brake_Power_Supply.kicad_pcb Brake_Power_Supply_panel.kicad_pcb
Fontconfig error: Cannot load default config file: No such file: (null)
Fontconfig error: Cannot load default config file: No such file: (null)
C:\Users\mchhe\OneDrive\Dokumente\KiCad\7.0\3rdparty\Python39\site-packages\shapely\set_operations.py:133: RuntimeWarning: invalid value encountered in intersection
  return lib.intersection(a, b, **kwargs)``

### Steps to Reproduce

Execute command above.
@yaqwsx if you need the PCB, i can send to your email. But not public here.

greenscreenflicker, 2023-05-09:
Thanks for clearifying this. I used another tool to export now, but I'd prefer KiKit. Please let me know once it works again.
Thank you for your support until now.

yaqwsx, 2023-05-30:
We enforced the Shapely version in the dependencies in 14bc40631a5ef781ca42f36ee1e011736d137d39.

closed 2023-05-30 by yaqwsx