yaqwsx/KiKit #519: KiKit destroyies tracks at panel borders

greenscreenflicker, 2023-03-06:
### Prerequisites

- [X] I have read [FAQ](https://yaqwsx.github.io/KiKit/latest/faq/)
- [X] I use KiKit at least version 1.3.0 (older version are not supported)

### KiKit version

kikit, version 1.3.0

### KiCAD version

6.0.11

### Operating system

Windows 11

### Description

Kitkit punches through a wire at panel side

See IN+ wire
[image]

Command:
kikit panelize  --layout "grid; rows: 4; cols: 5; space: 2mm"  --tabs "fixed; width: 20mm; vcount: 1; hcount: 1" --cuts "mousebites; drill: 0.5mm; spacing: 1mm; offset: 0.2mm; prolong: 0.5mm" --fiducials "3fid; hoffset: 5mm; voffset: 2.5mm; coppersize: 2mm; opening: 1mm;"  --text "simple; text: BLABLA, production run1; anchor: mt; voffset: 2.5mm; hjustify: center; vjustify: center;"  --framing "railstb; width: 5mm; space: 1mm;"  --post "millradius: 0.5mm"   --source "tolerance: 20mm"  DCC6024UT.kicad_pcb panel3.kicad_pcb

### Steps to Reproduce

execute command above

yaqwsx, 2023-03-07:
Positive mousebites offset insets the mousebites into the board. A negative one will push the mousebites into the tab. You might also want to specify precise tab location via annotations.

greenscreenflicker, 2023-03-07:
First: If someone has the same question how to specify offset: Look here and search for offset.
https://github.com/yaqwsx/KiKit/blob/master/docs/panelization/examples.md
That should fix the problem for now.

What should be the expected behavior?
The offset should be that way that no former PCB area is compromised. So a zero offset should mean, that no area to the board is added nor its area is reduced.

yaqwsx, 2023-03-07:
We currently follow the guideline of several PCB manufacturers for the convention of the offset. The reason for putting the mouse bites into the PCB is that when you break them, all burrs will be inside the intended PCB outline and thus, the PCB will fit the enclosure even when you don't deburr the PCB.

Changing the offset direction would be a breaking change (that would require a major release) and the current state is documented and explained, thus I have no plans for changing this.

closed 2023-03-07 by yaqwsx