yaqwsx/KiKit #525: Docker container tag should explicitly contain v6 or v7, no default

j4m3s, 2023-03-12:
### Prerequisites

- [X] I have read [FAQ](https://yaqwsx.github.io/KiKit/latest/faq/)
- [X] I use KiKit at least version 1.3.0 (older version are not supported)

### KiKit version

kikit, version 1.3.0

### KiCAD version

6 or 7 :)

### Operating system

Linux Ubuntu 20.04

### Description

Two separate docker container versions have been introduced, one with KiCad v6 installed and one with Kicad v7 installed.  

[ ] Why are there two versions of the docker container?  Given that the changes are backward compatible, and the user doesn't do anything in the container other than run kikit, why do you publish a version with Kicad 6?
[ ] If both images are required, why default to the version with KiCad 6? 
[ ] Again, assuming there's good reason to have both, this information should be added to the docs here: https://yaqwsx.github.io/KiKit/v1.3/installation/docker.  At the moment there's no way for people to know that there's a v7 image, and/or why they should use it.

As ever, I'm happy to submit a PR once decided :)

### Steps to Reproduce

n/a - documentation issue

yaqwsx, 2023-03-12:
- Once you build a panel with KiCAD v7, you cannot open it with KiCAD v6. This is why we support both versions.
- I didn't find it appropriate to migrate long-term users to KiCAD 7 (especially when we are less than one month after release); this is why we still default to the v6 container (because we released a **minor** revision of KiKit that should be backward compatible).
- I agree we should add this to the documentation.

j4m3s, 2023-03-12:
Ah interesting point re backward compatibility from 7 to 6.  Will 7 not open the v6 files transparently?

yaqwsx, 2023-03-12:
It will. But users using KiCAD 6 wouldn't be happy with panels produced by an image with KiCAD 7 as they wouldn't be able to open the resulting panel.

j4m3s, 2023-03-12:
Yes I see what you mean.  So really, users don't need the v7 docker image at the moment as everything "just works" with the v6 image?

sphawes, 2023-03-15:
Where is the kicad 7 docker image located? Hunted around the repo and wasnt able to find it.

In lieu of that docker image, I added the following to my Github CI yaml file (that uses the main docker image running Kicad 6.0) to install kicad 7 before running kikit and it worked well for me:

```
- name: Update to KiCad 7
  run: |
    add-apt-repository --yes ppa:kicad/kicad-7.0-releases
    apt -y update
    apt -y install --install-recommends kicad

also mega thanks @yaqwsx for making this tool! it's completely automated panelization for us, saved us countless hours over the past year or so. greatly appreciate the work that you do!

j4m3s, 2023-03-15:
Check the [docker hub](https://hub.docker.com/r/yaqwsx/kikit/tags) - there's a version of each of the latest images with a `-v7` suffix.  We need to add it to the docs but I'm keen to hear back from @yaqwsx on whether the v7's have any benefit over the v6's at the moment before I submit a PR.

j4m3s, 2023-03-15:
Ah understood. So users must choose the correct version to match their kicad install.

On that basis, I would recommend removing the "default" docker image and publish a `-v6` and a `-v7` version. We can make it very clear in the doc page that they should pull the correct version for their kicad version.

The reason for this suggestion is that it will avoid people pulling the default image, finding it doesn't work (because of a kicad version mismatch) and not knowing why.

I could envisage a lot of confused users thinking kikit is broken or creating corrupted output if not.

What do you think?

yaqwsx, 2023-03-16:
I am all for improving the documentation. If you submit a PR for that, I will be more than happy to accept it. I still prefer to preserve the default image until a new major release for not breaking existing stuff.

yaqwsx, 2024-03-16:
7a3ae3451850ad219072514894983ccb1cb83aaa

closed 2024-03-16 by yaqwsx