yaqwsx/KiKit #526: Including a subdir path in the output path for panelize should create the directory if it doesn't exist

j4m3s, 2023-03-12:
### Prerequisites

- [X] I have read the documentation and the proposed feature is not implemented.

### Description

Currently, if I run the command

```
kikit panelize \ 
    PCB.kicad_pcb \
    panels/PCB.panel.kicad_pcb
```

the command fails if the `panels` subdir doesn't already exist.  It would be nice if the directory could be created automatically, as it is with the output directory for `kikit fab`.

The reason for including a path in the panelization output is that it keeps things tidy when creating multiple panels, and allows the dir to be excluded from source control if required.

yaqwsx, 2023-03-12:
I am unsure if this is a wanted behavior or not. Consider the user invokes KiKit with `/same/long/path/to/the/output` instead of `/some/long/path/to/the/output`. That can lead to confusion. I am aware we do things differently in the `fab` command, but I am sure if that is a good thing.

What might be more appropriate is to add a flag (just like `mkdir` has the `-p` option) to control the behavior.

j4m3s, 2023-03-12:
You could output a message to the terminal when the new dir is created, that way if the file isn't where the user expected it to be, due to a typo, they would see in the terminal that a new dir was created to put them in.  You could do the same for fab too, so the user knows where to look if it's the first time they're running the command :)

yaqwsx, 2023-03-12:
Outputting a message is not much in the spirit of automation. When something probably unexpected should happen, we should fail. Only when we fail we should output anything (e.g., a hint to rerun the command with a flag).

j4m3s, 2023-03-12:
Yes understood, that's fair.  Equally though, in the spirit of automation, I'd expect the output directory to be created if it doesn't exist.  As it is, the following is printed to the console:
```
An error occurred: Unable to open newDir/output.kicad_pcb for reading.
No output files produced
```
Which is a bit confusing given that I wouldn't expect it to be *reading* the output file, it should be creating it.

In the spirit of [convention over configuration](https://en.wikipedia.org/wiki/Convention_over_configuration) the convention is to create output files/ paths if they don't exist, rather than requiring an additional configuration parameter be set.  That's what I expected with `kikit fab`, and that's what happens.  

All that said, it's your software so I'm totally happy if you close the issue as invalid and I'll write a shell script to wrap the call to kikit to create the directory if it doesn't exist.