yaqwsx/KiKit #527: JLCPCB component positioning incorrect after panelization

sityware, 2023-03-15:
### Prerequisites

- [X] I have read [FAQ](https://yaqwsx.github.io/KiKit/latest/faq/)
- [X] I use KiKit at least version 1.3.0 (older version are not supported)

### KiKit version

kikit, version 1.3.0

### KiCAD version

6.0.10

### Operating system

Windows 11

### Description

After panelization and exporting Gerbers and POS files. When adding Gerbers to JLCPCB the component position is incorrect.

I have found a solution if I edit the .PCB file and remove this line 
`    (aux_axis_origin 56.5 20)`

JLCPCB now has my components in the correct position.

This is my KiKit config if it helps

`{
    "layout": {
        "hspace": "2mm",
        "vspace": "2mm",
        "rows": "3",
        "cols": "3"
    },
    "tabs": {
        "type": "fixed",
        "vwidth": "5mm",
        "hwidth": "5mm",
        "vcount": "2",
        "hcount": "2"
    },
    "cuts": {
        "type": "mousebites"
    },
    "framing": {
        "type": "railstb",
        "vspace": "3mm",
        "width": "8mm"
    },
    "tooling": {
        "type": "3hole",
        "hoffset": "2.5mm",
        "voffset": "2.5mm",
        "size": "1.152mm",
        "paste": "True"
    },
    "fiducials": {
        "type": "3fid",
        "hoffset": "20mm",
        "voffset": "4.2mm",
        "opening": "2mm"
    },
    "text": {
        "type": "simple",
        "hoffset": "-9mm",
        "voffset": "-2.8mm",
        "width": "2mm",
        "height": "2mm",
        "thickness": "0.4mm",
        "hjustify": "right",
        "vjustify": "right",
        "text": "TEST -->",
        "anchor": "br"
    },
    "post": {
        "millradius": "1mm"
    }
}`

### Steps to Reproduce

Panelise using KiKit.
Import to JLCPCB

yaqwsx, 2023-03-16:
How do you export the POS file for JLC PCB?

sityware, 2023-03-16:
Apologies, seems like the default setting on the POS dialog has Use Drill / Place file origin ticked.
After I disabled this the position is fine.

closed 2023-03-16 by sityware