yaqwsx/KiKit #531: "Kikit separate" command creates Edge.cuts that doesn't exist in the source file.

Romly-Romly, 2023-03-22:
### Prerequisites

- [X] I have read [FAQ](https://yaqwsx.github.io/KiKit/latest/faq/)
- [X] I use KiKit at least version 1.3.0 (older version are not supported)

### KiKit version

1.3.0

### KiCAD version

7.0.0

### Operating system

Windows 10

### Description

Thank you for creating a wonderful tool.
I'm completely new to Github and writing an issue, sorry if I'm breaking any community guidelines. (Also, I'm not good at English since it's my learning language.)

So, I used the `kikit separate` command to separate my kicad_pcb file which contains multiple PCBs.
The output file was created successfully, but it has weird arc line that is not exist in the original file.
I thought it can be a bug so I'm writing this issue just in case.

Here is the screenshot, the left is the original and the right is the file created by Kikit. You can see the center hole shape has changed. The lines are in Edge.cuts layer.
[image]

I'm attaching a ZIP file which contains the original kicad_pcb and the output one. (The original file only contains a single PCB, I deleted others so it'd be simple to point the issue)
[kicad_pcb_files.zip](https://github.com/yaqwsx/KiKit/files/11036219/kicad_pcb_files.zip)

### Steps to Reproduce

The command I used was this:
`kikit separate --source "rectangle; tlx: -40mm; tly: -100mm; brx:40mm; bry:10mm" stepper_board.kicad_pcb output.kicad_pcb`

yaqwsx, 2023-03-22:
This is a duplicate #453. See the thread for a work-around.

yaqwsx, 2023-06-27:
Closing in favor of #453. We plan to solve this by migrating to different polygon engine, but this will take some time.

closed 2023-06-27 by yaqwsx

yaqwsx, 2024-03-06:
As the switch to a new geometry engine will take some time, I tried to fix the current arc welder in a2c9891. The upstream version should fix your issue.