yaqwsx/KiKit #536: Panelize command

ludiazv, 2023-04-02:
### Prerequisites

- [X] I have read [FAQ](https://yaqwsx.github.io/KiKit/latest/faq/)
- [X] I use KiKit at least version 1.3.0 (older version are not supported)

### KiKit version

kikit, version 1.3.0

### KiCAD version

7.0.1

### Operating system

docker 1.3.0-v7

### Description

Thanks for this great tool.

```panel``` command does not translated net assignments(patterns) from the source board. The operation does not fail but if a subsequent ```fab``` commands fail if DRC check is on.

This is the section of section of the **source** ```.pro``` file:
````
 "net_settings": {
    "classes": [
      {
        "bus_width": 12,
        "clearance": 0.2,
        "diff_pair_gap": 0.25,
        "diff_pair_via_gap": 0.25,
        "diff_pair_width": 0.2,
        "line_style": 0,
        "microvia_diameter": 0.3,
        "microvia_drill": 0.1,
        "name": "Default",
        "pcb_color": "rgba(0, 0, 0, 0.000)",
        "schematic_color": "rgba(0, 0, 0, 0.000)",
        "track_width": 0.25,
        "via_diameter": 0.8,
        "via_drill": 0.4,
        "wire_width": 6
      },
      {
        "bus_width": 12,
        "clearance": 0.127,
        "diff_pair_gap": 0.25,
        "diff_pair_via_gap": 0.25,
        "diff_pair_width": 0.2,
        "line_style": 0,
        "microvia_diameter": 0.3,
        "microvia_drill": 0.1,
        "name": "NC",
        "pcb_color": "rgba(0, 0, 0, 0.000)",
        "schematic_color": "rgba(0, 0, 0, 0.000)",
        "track_width": 0.13,
        "via_diameter": 0.8,
        "via_drill": 0.4,
        "wire_width": 6
      },
      {
        "bus_width": 12,
        "clearance": 0.2,
        "diff_pair_gap": 0.25,
        "diff_pair_via_gap": 0.25,
        "diff_pair_width": 0.2,
        "line_style": 0,
        "microvia_diameter": 0.3,
        "microvia_drill": 0.1,
        "name": "PWR",
        "pcb_color": "rgba(0, 0, 0, 0.000)",
        "schematic_color": "rgba(0, 0, 0, 0.000)",
        "track_width": 0.25,
        "via_diameter": 0.8,
        "via_drill": 0.4,
        "wire_width": 6
      },
      {
        "bus_width": 12,
        "clearance": 0.15,
        "diff_pair_gap": 0.25,
        "diff_pair_via_gap": 0.25,
        "diff_pair_width": 0.2,
        "line_style": 0,
        "microvia_diameter": 0.3,
        "microvia_drill": 0.1,
        "name": "signal",
        "pcb_color": "rgba(0, 0, 0, 0.000)",
        "schematic_color": "rgba(0, 0, 0, 0.000)",
        "track_width": 0.16,
        "via_diameter": 0.5,
        "via_drill": 0.3,
        "wire_width": 6
      }
    ],
    "meta": {
      "version": 3
    },
    "net_colors": null,
    "netclass_assignments": null,
    "netclass_patterns": [
      {
        "netclass": "PWR",
        "pattern": "+3.3V"
      },
      {
        "netclass": "PWR",
        "pattern": "/3V3R"
      },
      {
        "netclass": "PWR",
        "pattern": "GND"
      },
      {
        "netclass": "signal",
        "pattern": "/BOOT"
      },
      {
        "netclass": "signal",
        "pattern": "/CS"
      },
      {
        "netclass": "signal",
        "pattern": "/D+"
      },
      {
        "netclass": "signal",
        "pattern": "/D-"
      },
      {
        "netclass": "signal",
        "pattern": "/D0"
      },
      {
        "netclass": "signal",
        "pattern": "/D1"
      },
      {
        "netclass": "signal",
        "pattern": "/G0"
      },
      {
        "netclass": "signal",
        "pattern": "/MISO"
      },
      {
        "netclass": "signal",
        "pattern": "/MOSI"
      },
      {
        "netclass": "signal",
        "pattern": "/RST"
      },
      {
        "netclass": "signal",
        "pattern": "/RX"
      },
      {
        "netclass": "signal",
        "pattern": "/SCK"
      },
      {
        "netclass": "signal",
        "pattern": "/SCL"
      },
      {
        "netclass": "signal",
        "pattern": "/SDA"
      },
      {
        "netclass": "signal",
        "pattern": "/TX"
      },
      {
        "netclass": "signal",
        "pattern": "Net-(D1-Pad2)"
      },
      {
        "netclass": "signal",
        "pattern": "Net-(JP1-Pad1)"
      },
      {
        "netclass": "NC",
        "pattern": "unconnected-*"
      }
    ]

````

But in the panel ```.pro``` no assigments are made:

````
.....

{
        "bus_width": 12,
        "clearance": 0.127,
        "diff_pair_gap": 0.25,
        "diff_pair_via_gap": 0.25,
        "diff_pair_width": 0.2,
        "line_style": 0,
        "microvia_diameter": 0.3,
        "microvia_drill": 0.1,
        "name": "C3M2_5-NC",
        "pcb_color": "rgba(0, 0, 0, 0.000)",
        "schematic_color": "rgba(0, 0, 0, 0.000)",
        "track_width": 0.13,
        "via_diameter": 0.8,
        "via_drill": 0.4,
        "wire_width": 6,
        "nets": []
      },
      {
        "bus_width": 12,
        "clearance": 0.2,
        "diff_pair_gap": 0.25,
        "diff_pair_via_gap": 0.25,
        "diff_pair_width": 0.2,
        "line_style": 0,
        "microvia_diameter": 0.3,
        "microvia_drill": 0.1,
        "name": "C3M2_5-PWR",
        "pcb_color": "rgba(0, 0, 0, 0.000)",
        "schematic_color": "rgba(0, 0, 0, 0.000)",
        "track_width": 0.25,
        "via_diameter": 0.8,
        "via_drill": 0.4,
        "wire_width": 6,
        "nets": []
      },
      {
        "bus_width": 12,
        "clearance": 0.15,
        "diff_pair_gap": 0.25,
        "diff_pair_via_gap": 0.25,
        "diff_pair_width": 0.2,
        "line_style": 0,
        "microvia_diameter": 0.3,
        "microvia_drill": 0.1,
        "name": "C3M2_5-signal",
        "pcb_color": "rgba(0, 0, 0, 0.000)",
        "schematic_color": "rgba(0, 0, 0, 0.000)",
        "track_width": 0.16,
        "via_diameter": 0.5,
        "via_drill": 0.3,
        "wire_width": 6,
        "nets": []
      }
    ],
    "meta": {
      "version": 3
    },
    "net_colors": null,
    "netclass_assignments": null,
    "netclass_patterns": []

````

### Steps to Reproduce

run panel command with net assignments.

yaqwsx, 2023-06-27:
This should be fixed in 119665a37c019fcaf5479f8e8c3be077688e6971.

closed 2023-06-27 by yaqwsx

yaqwsx, 2023-06-27:
Of course, wait for the nightly container to rebuild. Or build it yourself if you are in a hurry.

ludiazv, 2023-07-11:
I checked it with nightly docker. It works! now the net patterns are translated in panelize.

Many thanks for support!