yaqwsx/KiKit #539: Mouse Bites will not generate on a curve

cobra18t, 2023-04-15:
### Prerequisites

- [X] I have read [FAQ](https://yaqwsx.github.io/KiKit/latest/faq/)
- [X] I use KiKit at least version 1.3.0 (older version are not supported)

### KiKit version

kikit, version 1.3.0

### KiCAD version

7.0

### Operating system

Linux Mint 21.1

### Description

While mousebites are most effective on a straight edge, they should also populate on a curved outline.

Here you can see they are only generated on the straight edges. I would have expected them to also follow the curves
[image]

### Steps to Reproduce

Use a circular board and try to generate a panel with mousebites.

yaqwsx, 2023-04-15:
Could you provide **precise** steps to reproduce (just like the issue template asks for)? That is including the exact command you issue and the particular board you use? Mouse bites on arbitrary shapes seems to work just fine on any case I tried:

[image]

cobra18t, 2023-04-15:
Here is a test file:
[test.kicad_pcb.zip](https://github.com/yaqwsx/KiKit/files/11238562/test.kicad_pcb.zip)

Preset:
```
{
    "layout": {
        "alternation": "rows",
        "hspace": "1.5mm",
        "vspace": "0.5mm",
        "rows": "4",
        "cols": "3"
    },
    "tabs": {
        "type": "fixed"
    },
    "cuts": {
        "type": "mousebites"
    }
}
```
Result:
[image]

yaqwsx, 2023-04-15:
I still struggle to reproduce your issue, everything works in my tests:

[image]

Could you change the style of cuts to line (https://yaqwsx.github.io/KiKit/v1.3/panelization/cli/#layer) and see if all cuts are generated correctly?

cobra18t, 2023-04-15:
Yes, "layer" works perfectly.

yaqwsx, 2023-04-15:
What version of Shapely do you have installed on your system?

cobra18t, 2023-04-15:
shapely   2.0.1

yaqwsx, 2023-04-15:
Can you try it with v1.8.5?

cobra18t, 2023-04-15:
Could you remind me how to revert to a specific version?

cobra18t, 2023-04-15:
Nevermind, just found it. Standby for test.

cobra18t, 2023-04-15:
Yes. That solves it!

closed 2023-05-09 by yaqwsx