yaqwsx/KiKit #542: KiKit panelize fails for KiCad 7.0.2 (when using ZONEs)

set-soft, 2023-04-18:
### Prerequisites

- [X] I have read [FAQ](https://yaqwsx.github.io/KiKit/latest/faq/)
- [X] I use KiKit at least version 1.3.0 (older version are not supported)

### KiKit version

1.3.0-6

### KiCAD version

7.0.2

### Operating system

Debian GNU/Linux 11.6

### Description

Unfortunately KiCad developers deleted an BOARD_ITEM.Duplicate() wrapper.

Error introduced by https://gitlab.com/kicad/code/kicad/-/commit/8e579ec8e0e8d485645d18cb5d897fc2174307c5

Fixed by https://gitlab.com/kicad/code/kicad/-/commit/7843e6a70904e9960114a0dc6cfff816b27c116f

But this error is in 7.0.2 KiCad tag, so you'll get a crash when trying to use a ZONE.Duplicate()

I'm patching `pcbnew` in my docker images using:  https://github.com/set-soft/pcbnewTransition/commit/fcc76c9e52354b559f08548f298e956208114081
It has an extra cast that I removed in https://github.com/set-soft/pcbnewTransition/commit/b5e83eef114cbe60077e4fbbc58232025967927d
But it was there for 8 years

So I recommend applying the same or equivalent patch to  pcbnewTransition and doing a new release.

### Steps to Reproduce

Just run panelize on any board with ZONEs

JeremiahGillis, 2023-04-21:
I ran into this same issue today on two of my projects. It would be great to get a fix until KiCad updates 7.0.

set-soft, 2023-04-21:
Hi @JeremiahGillis !
You can try using the pcbnewTransition from this repo: https://github.com/set-soft/pcbnewTransition

wburchard, 2023-05-04:
I've also experienced the same thing.  I was using KiKat one day, then the next it refuses to work.

Ubuntu 22.10, KiCad Application: KiCad x86_64 on x86_64

Version: 7.0.2-6a45011f42~172~ubuntu22.10.1, release build

Libraries:
	wxWidgets 3.2.1
	FreeType 2.12.1
	HarfBuzz 2.7.4
	FontConfig 2.13.1
	libcurl/7.85.0 OpenSSL/3.0.5 zlib/1.2.11 brotli/1.0.9 zstd/1.5.2 libidn2/2.3.3 libpsl/0.21.0 (+libidn2/2.3.2) libssh/0.9.6/openssl/zlib nghttp2/1.49.0 librtmp/2.3

Platform: Ubuntu 22.10, 64 bit, Little endian, wxGTK, mate, x11

Build Info:
	Date: Apr 17 2023 07:57:55
	wxWidgets: 3.2.1 (wchar_t,wx containers) GTK+ 3.24
	Boost: 1.74.0
	OCC: 7.6.3
	Curl: 7.85.0
	ngspice: 38
	Compiler: GCC 12.2.0 with C++ ABI 1017

Build settings:
	KICAD_SPICE=ON

How I reproduce the error:

1. Make sure all instances of KiCad have been closed.
2. Create a new KiCad project.
3. Open PcbNew.
4. Open KiKit
5. Select an existing Pcb file.
6. Click Panelize
7. Running kitkit pop-up appears, stays there until it, as well as KiCad closes.

While noticing that that kitkit version 1.3.0 is now available, I've been unable to get it installed.  I can uninstall it then reinstall it, but I keep getting version 1.1.0.

[image]

set-soft, 2023-05-04:
> While noticing that that kitkit version 1.3.0 is now available, I've been unable to get it installed. I can uninstall it then reinstall it, but I keep getting version 1.1.0.

Are you sure you didn't install it using `pip`? (not only using the Plug-in manager?)

You can also try using the KiBot repo: https://set-soft.github.io/debian/

It has KiKit 1.3.0 Debian packages and also a pcbnewTransition package that fixes the ZONEs issue.

wburchard, 2023-05-04:
I uninstalled KiKit, then installed the deb package. Version 1.3.0 now appears:

[image]

KiKit does not show as installed in the PCM:

[image]

Unfortunately, KiKit still crashes, taking KiCad down with it, when I attempt to panelize a board.

ERROR: apport (pid 7296) Thu May  4 16:10:51 2023: called for pid 7196, signal 11, core limit 0, dump mode 1
ERROR: apport (pid 7296) Thu May  4 16:10:51 2023: executable: /usr/bin/kicad (command line "kicad")
ERROR: apport (pid 7296) Thu May  4 16:10:52 2023: debug: session gdbus call: (true,)

ERROR: apport (pid 7296) Thu May  4 16:11:09 2023: wrote report /var/crash/_usr_bin_kicad.1000.crash
ERROR: apport (pid 8126) Thu May  4 16:13:58 2023: called for pid 7983, signal 11, core limit 0, dump mode 1
ERROR: apport (pid 8126) Thu May  4 16:13:58 2023: executable: /usr/bin/kicad (command line "kicad")
ERROR: apport (pid 8126) Thu May  4 16:13:59 2023: debug: session gdbus call: (true,)

ERROR: apport (pid 8126) Thu May  4 16:14:16 2023: wrote report /var/crash/_usr_bin_kicad.1000.crash

Unfortunately, the crash files are not in the /var/crash/ folder (possibly deleted by KiCad).

Is there any additional information I can provide that would help?

set-soft, 2023-05-05:
Hi @wburchard !

> I uninstalled KiKit, then installed the deb package. Version 1.3.0 now appears:

Good.

> KiKit does not show as installed in the PCM:

This is logic. The PCM manages a special directory dedicated to the plug-ins downloaded with it.
The Debian package installs it at system level, allowing all users in your machine to use it, no need for a copy for each user. The only drawback is that KiKit libs won't be automagically discovered, you must explicitly tell KiCad they exist. No need to copy, they are there, is just that KiCad was designed in a very "Windows" style (oriented for 1 user).

> Unfortunately, KiKit still crashes, taking KiCad down with it, when I attempt to panelize a board.

First thing to check is that the new pcbnewTransition is there, try running `dpkg -l python3-pcbnewtransition` the package should be installed and with version 0.3.4-4.

> ERROR: apport (pid 7296) Thu May 4 16:10:51 2023: called for pid 7196, signal 11, core limit 0, dump mode 1 ERROR: apport (pid 7296) Thu May 4 16:10:51 2023: executable: /usr/bin/kicad (command line [...]

Don't worry, this won't help.
Can you try running KiKit from the command line?  Look in the dialog, KiKit shows the command line. But before doing it  enable *debug* / *trace*
This will show where KiKit failed.

> 
> Is there any additional information I can provide that would help?

wburchard, 2023-05-05:
It appears that python3-pcbnewtransition is installed:

[image]

I finally figured out I had to use the Python Scripting Console to get trace information:

Py 0.9.8
Python 3.10.7 (main, Mar 10 2023, 10:47:39) [GCC 12.2.0] on linux
Type "help", "copyright", "credits" or "license" for more information.
Startup script executed: /home/william/.config/kicad/7.0/PyShell_pcbnew_startup.py

kikit panelize \
    --layout 'rows: 3' \
    --cuts 'type: mousebites' \
    --framing 'type: railstb' \
    --debug 'trace: True' \
    /home/william/KicadProjects/TPS5100_1S_LipoCharger/TPS5100_1S_LipoCharger.kicad_pcb panel.kicad_pcb
Traceback (most recent call last):
  File "/usr/lib/python3.10/code.py", line 63, in runsource
    code = self.compile(source, filename, symbol)
  File "/usr/lib/python3.10/codeop.py", line 153, in __call__
    return _maybe_compile(self.compiler, source, filename, symbol)
  File "/usr/lib/python3.10/codeop.py", line 70, in _maybe_compile
    compiler(source + "\n", filename, symbol)
  File "/usr/lib/python3.10/codeop.py", line 118, in __call__
    codeob = compile(source, filename, symbol, self.flags, True)
  File "<input>", line 1
    kikit(panelize \)
                    ^
SyntaxError: unexpected character after line continuation character

I'm not really familiar with Python, but after staring at it a while the '\)' following panelize jumped out at me, so tried the following:

kikit panelize --layout 'rows: 3' --cuts 'type: mousebites' --framing 'type: railstb' debug 'trace: True' '/home/william/KicadProjects/TPS5100_1S_LipoCharger/TPS5100_1S_LipoCharger.kicad_pcb' 'panel.kicad_pcb'
Traceback (most recent call last):
  File "/usr/lib/python3.10/code.py", line 63, in runsource
    code = self.compile(source, filename, symbol)
  File "/usr/lib/python3.10/codeop.py", line 153, in __call__
    return _maybe_compile(self.compiler, source, filename, symbol)
  File "/usr/lib/python3.10/codeop.py", line 70, in _maybe_compile
    compiler(source + "\n", filename, symbol)
  File "/usr/lib/python3.10/codeop.py", line 118, in __call__
    codeob = compile(source, filename, symbol, self.flags, True)
  File "<input>", line 1
    kikit(panelize --layout 'rows: 3' --cuts 'type: mousebites' --framing 'type: railstb' debug 'trace: True' '/home/william/KicadProjects/TPS5100_1S_LipoCharger/TPS5100_1S_LipoCharger.kicad_pcb' 'panel.kicad_pcb')
          ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
SyntaxError: invalid syntax. Perhaps you forgot a comma?

This is probably pretty simple, but I'm stumped.

set-soft, 2023-05-08:
> It appears that python3-pcbnewtransition is installed:

Ok.

> I finally figured out I had to use the Python Scripting Console to get trace information:

No, you must use a regular console, run the kikit command using the shell, like any other Linux command.

`kikit` is just a command, not a Python function. The fact that is implemented y Python is another story ;-)

Just paste the kikit command in the same console you executed *dpkg -l python3-pcbnewtransition*

wburchard, 2023-05-08:
[image]
[image]

Please see above images.  I do not see any trace output.  Is it written someplace?  This is why I tried the scripting console.

set-soft, 2023-05-09:
Hi @wburchard !
Try loading panel.kicad_pcb with KiCad, if this is OK, then KiKit worked fine and you have what you want!

yaqwsx, 2023-05-30:
KiCAD 7.0.5 was released which fixes this regression.

closed 2023-05-30 by yaqwsx