yaqwsx/KiKit #544: An error occurred: <built-in function EXCELLON_WRITER_CreateDrillandMapFilesSet> returned a result with an error set

greenscreenflicker, 2023-05-03:
### Prerequisites

- [X] I have read [FAQ](https://yaqwsx.github.io/KiKit/latest/faq/)
- [X] I use KiKit at least version 1.3.0 (older version are not supported)

### KiKit version

kikit, version 1.3.0+16.gf84d28c

### KiCAD version

7.02

### Operating system

Windows

### Description

Executing the following command gives the following error:
```kikit fab jlcpcb --no-drc --assembly --schematic GPS-Flyback.kicad_sch GPS-Flyback.kicad_pcb gerber/
Fontconfig error: Cannot load default config file: No such file: (null)
Fontconfig error: Cannot load default config file: No such file: (null)
An error occurred: <built-in function EXCELLON_WRITER_CreateDrillandMapFilesSet> returned a result with an error set
No output files produced

### Steps to Reproduce

execute code above. if you need pcb, i can mail you privately.

greenscreenflicker, 2023-05-03:
@yaqwsx just tried with another pcb - there it works fine. So it must be something drill specific, judging from the error msg.

yaqwsx, 2023-05-09:
Hi! I believe this is a bug in KiCAD (based on the error message). Could you e-mail me the board to email@honzamrazek.cz?

yaqwsx, 2023-05-30:
I managed to locate the source of the error. KiCAD changed the way it handles logging (which caused this error). Fixed in 76643105e8969e22f9edca11b57082dde5a805f4.