yaqwsx/KiKit #549: KiKit panelize fails for KiCad 7.0.3rc1

set-soft, 2023-05-11:
### Prerequisites

- [X] I have read [FAQ](https://yaqwsx.github.io/KiKit/latest/faq/)
- [X] I use KiKit at least version 1.3.0 (older version are not supported)

### KiKit version

1.3.0-7

### KiCAD version

7.0.3rc1

### Operating system

Debian GNU/Linux 11.7

### Description

This is just informative because:
1. KiCad 7.0.3 isn't yet released, is scheduled for this Sunday.
2. I already proposed an MR to KiCad repo (https://gitlab.com/kicad/code/kicad/-/merge_requests/1605)

But I think is important to pay attention to this issue and be prepared in case the MR is rejected.

The problem is in KiCad 7.0.3 API, Jeff fixed some calls to **GetShownText** that unintentionally asked for extra text. To avoid it he changed *aAllowExtraText* from optional to mandatory. But this was the second argument to the functions, so he was forced to exchange its place with the first argument. So now all **GetShownText** members are completely incompatible:
1. They have one mandatory argument
2. The argument order is inverted

KiKit panelize calls **GetShownText** without arguments, and hence breaks.

I proposed a patch that fixes the first change (no arguments) and respects the second (arguments inversion).

I hope the change is accepted, otherwise KiKit will have problems with the last two releases.

### Steps to Reproduce

Just any panel that uses text.

set-soft, 2023-05-12:
MR aproved, don't worry.

closed 2023-05-12 by set-soft

set-soft, 2023-05-15:
None that I'm aware off.
I only found 7.99 nightlies.
I'm asking about it in the open dev list.
I just compile 7.0 branch manually from time to time.

yaqwsx, 2023-05-15:
Compiling it is an option, but I would like to avoid it (as it makes the CI longer and also adds extra points of failure and maintenance). Thanks for asking!

set-soft, 2023-05-15:
Yes, I do it locally, at least on GitLab the KiCad CI/CD pipelines (compile+tests) takes almost 4 hours.
GitLab and the PPA repos gives some privileges to KiCad project so they can use more than 10 times the quota. I don't know if they pay for it.

set-soft, 2023-05-16:
> are there nightly builds of Debian packages of the v7 branch?

https://launchpad.net/~kicad/+archive/ubuntu/kicad-7.0-nightly

Ian McInerney told me: *the terminology we prefer to use is "testing builds" for builds using the tip of the stable branch and "nightly builds" for builds using the tip of the master branch.*

yaqwsx, 2023-05-16:
I watch the mailing list so I added the tests this morning: https://github.com/yaqwsx/KiKit/commit/6bf9848b966a14db1859a7a3eeb6c757dde42179