yaqwsx/KiKit #550: Unsupported violation object count

holmesrichards, 2023-05-18:
### Prerequisites

- [X] I have read [FAQ](https://yaqwsx.github.io/KiKit/latest/faq/)
- [X] I use KiKit at least version 1.3.0 (older version are not supported)

### KiKit version

kikit, version 1.3.0

### KiCAD version

Application: KiCad x86_64 on x86_64  Version: 7.0.2-6a45011f42~172~ubuntu22.04.1, release build  Libraries: 	wxWidgets 3.2.1 	FreeType 2.11.1 	HarfBuzz 6.0.0 	FontConfig 2.13.1 	libcurl/7.81.0 OpenSSL/3.0.2 zlib/1.2.11 brotli/1.0.9 zstd/1.4.8 libidn2/2.3.2 libpsl/0.21.0 (+libidn2/2.3.2) libssh/0.9.6/openssl/zlib nghttp2/1.43.0 librtmp/2.3 OpenLDAP/2.5.14  Platform: Linux Mint 21.1, 64 bit, Little endian, wxGTK, cinnamon, x11  Build Info: 	Date: Apr 17 2023 07:58:00 	wxWidgets: 3.2.1 (wchar_t,wx containers) GTK+ 3.24 	Boost: 1.74.0 	OCC: 7.5.2 	Curl: 7.88.1 	ngspice: 38 	Compiler: GCC 11.3.0 with C++ ABI 1016  Build settings: 	KICAD_SPICE=ON

### Operating system

Linux Mint 21.1 Cinnamon

### Description

kikit Gerber generation fails with error (for some projects):

An error occurred: Unsupported violation object count
No output files produced

Note kicad-cli pcb export gerbers runs with no errors (for the same projects).

### Steps to Reproduce

Project: [pmf.zip](https://github.com/yaqwsx/KiKit/files/11508174/pmf.zip)

If I separate boards
```
kikit separate --source 'annotation; ref: B1' 'pmf.kicad_pcb' 'pmf_B1.kicad_pcb'
kikit separate --source 'annotation; ref: B2' 'pmf.kicad_pcb' 'pmf_B2.kicad_pcb'
```
and then generate Gerbers
```
kikit fab jlcpcb pmf_B1.kicad_pcb ./Gerbdir
kikit fab jlcpcb pmf_B2.kicad_pcb ./Gerbdir
```
I get the above error for B1 but not for B2.

If I generate Gerbers using the original PCB file (unseparated) I also get the above error.

morio, 2023-05-24:
have you tried kikit fab jlcpcb --no-drc [rest of command] ?

That sometimes works for me when I get that error, haven't figured why yet.

holmesrichards, 2023-05-24:
That does indeed prevent the error, but I don't really want to skip DRC. I can and usually do run DRC anyway but catching it when I don't is part of the point here.

Anyway, your remark led me to further information. When I run DRC on that project I get no errors, but I do get two warnings. ("Copper sliver", whose meaning I don't even understand, but it seems to be pointing to places on the ground fill that look acceptable to me and in any case I don't see how to change it; it's just how KiCad filled in around that pad.) If I edit violation severity for "copper sliver" to ignore instead of warning, then `kikit fab` runs without errors.

So apparently kikit is failing — and giving a very cryptic message — when DRC throws a warning? But I don't think it used to do that, and I don't think it should: Errors should lead to an abort, warnings should not.