yaqwsx/KiKit #551: Possibility to offset the alternated PCB's in a row/column.

TweetyDaBird, 2023-05-24:
### Prerequisites

- [X] I have read the documentation and the proposed feature is not implemented.

### Description

I just happen to be working with triangle shaped PCB's for once, and it would save tremendous space on the panel, if the PCB's could be offset to overlap the tip part.

yaqwsx, 2023-05-24:
You can write a Layout plugin for that: https://yaqwsx.github.io/KiKit/v1.3/panelization/plugins/

set-soft, 2023-05-24:
> Thanks though, and I might actually do it one day if I feel masochistic enough.

You could ask @yaqwsx how much it will charge you to solve your problem.

whinis, 2023-08-02:
So I do a bit of hardware and software and have a need for this. I wrote a very simple plugin that simply takes an offset in the form of rowoffset and coloffset that I placed in layout as seen below
```
"layout": {
        "hspace": "2.2mm",
        "vspace": "10mm",
        "rowoffset":"42mm",
        "coloffset":"7mm",
        "rows": "4",
        "cols": "2",
        "alternation":"cols",
        "type":"plugin",
        "code":"staggered_layout.py.StaggeredLayout"
    },
```
    However, if I change my offset I tend to get no tabs/mouse bites to render except those near the frame and get the following error
    ```
shapely/set_operations.py:133: RuntimeWarning: invalid value encountered in intersection
  return lib.intersection(a, b, **kwargs)
```
As I increase or decrease the offset sometimes some tabs internally will render on one side of the board but not the other side of the board. Image example given. Do I need to also make a custom tab annotation ?
![image](https://github.com/yaqwsx/KiKit/assets/7612064/a7800673-7bc8-4b5e-9964-bfbda37d9c09)

whinis, 2026-06-13:
3 years later and I still have this issue. Any update on why the tabs are not connecting? Attached is the tab locations and how its affecting a different board

[image]

[image]