yaqwsx/KiKit #554: Fatal/Urgent: Panalize introduces new tracks generating shorts.

greenscreenflicker, 2023-06-07:
### Prerequisites

- [X] I have read [FAQ](https://yaqwsx.github.io/KiKit/latest/faq/)
- [X] I use KiKit at least version 1.3.0 (older version are not supported)

### KiKit version

kikit, version 1.3.0+16.gf84d28c

### KiCAD version

7.0.5

### Operating system

Windows 11

### Description

Please have a look at the attached pcb.   [WCH_LINK_EADAPTER.zip](https://github.com/yaqwsx/KiKit/files/11673555/WCH_LINK_EADAPTER.zip)

The sources are in the base folder, while the panel is generated via kikit. The kikit commands can be found in make_gerbers.

The shorts are shown below.

[image]
[image]

### Steps to Reproduce

see above

greenscreenflicker, 2023-06-07:
Ups, you are right.
Is there an option to add a warning to avoid that mistake?

yaqwsx, 2023-06-07:
Run DRC after panelization.

greenscreenflicker, 2023-06-07:
Good (and obviously good) proposal.

But there is a real-life-issue with that.
I have some faults that I manually exclude. Are those exclusions also copied?

greenscreenflicker, 2023-06-07:
Ok. Thank you for your support.

closed 2023-06-07 by greenscreenflicker