yaqwsx/KiKit #555: Crash

greenscreenflicker, 2023-06-07:
### Prerequisites

- [X] I have read [FAQ](https://yaqwsx.github.io/KiKit/latest/faq/)
- [X] I use KiKit at least version 1.3.0 (older version are not supported)

### KiKit version

kikit, version 1.3.0+26.g56c696a

### KiCAD version

7.0.5

### Operating system

Windows 11

### Description

I get the following error. The pcb was send to email@...
[image]

### Steps to Reproduce

Execute the following command

mkdir panel
cd panel

kikit fab jlcpcb --assembly --schematic ../GPS-Flyback.kicad_sch ../GPS-Flyback.kicad_pcb gerber_single/

yaqwsx, 2023-06-07:
This is (known) bug in KiCAD. The error is not fatal, you can press Cancel and continue. These messages are suppressed on Linux and Mac. Once I get on a Windows computer, I can try also suppressing them for Windows.

greenscreenflicker, 2023-06-07:
Ok. Thanks. I'll leave the bug open. But I edited the title.

yaqwsx, 2023-11-10:
Please try the newest version. bc1c9a5 should address this.

greenscreenflicker, 2023-11-10:
If i will see this in the future, I'll report again. Thanks for fixing @yaqwsx

closed 2023-11-12 by yaqwsx