yaqwsx/KiKit #557: Feature Request: Crop Silkscreen going outside of PCBs area.

greenscreenflicker, 2023-06-07:
### Prerequisites

- [X] I have read the documentation and the proposed feature is not implemented.

### Description

I have lot's of connectors standing outside of the PCB.

For example:
[image]

I would like to panilze them also above and below. But I don't want the silkscreen from the other PCB on it.
Could there be a feature implemented, cropping that silkscreen outside of PCB?

greenscreenflicker, 2023-07-09:
@yaqwsx ping

yaqwsx, 2023-07-23:
The feature is straightforward but quite laborious to implement without a proper geometry library (as we have to edit the silkscreen geometry). At the moment, we use Shapely which isn't a good candidate for it.

However, I plan to migrate KiKit to a geometry library soon which should make this feature much easier to implement.