yaqwsx/KiKit #559: GUI panelization fails when PCB filename contains spaces

therealergo, 2023-06-11:
### Prerequisites

- [X] I have read [FAQ](https://yaqwsx.github.io/KiKit/latest/faq/)
- [X] I use KiKit at least version 1.3.0 (older version are not supported)

### KiKit version

kikit, version 1.3.0

### KiCAD version

7.0.5

### Operating system

Windows 10 64-bit

### Description

Panelizing a board whose name contains a space from the GUI panelization tool causes KiCAD to crash. This does not crash and works as expected on older versions of KiKit-- I'm not sure what version I upgraded from, but it was probably 1.1.x.

From the "KiKit CLI command" window generated, it looks like the path/filename isn't being quoted, even though it contains a space. Running the command that is generated using the KiCAD command prompt gives the following error:
```
Usage: kikit panelize [OPTIONS] INPUT OUTPUT
Try 'kikit panelize --help' for help.

Error: Got unexpected extra argument (panel.kicad_pcb)
```
Quoting the filename in the command or removing the space from the input filename fixes the issue.

### Steps to Reproduce

1. Select a PCB for "Input File" whose filename contains a space.
2. Add any panelization options, e.g. set Framing to "type: frame".
3. Click the "Panelize" button.

yaqwsx, 2023-06-12:
A space shouldn't cause this - no shell is invoked when penalizing directly within GUI. Can you confirm that the same PCB without a space in a name panelizes just fine?

It is possible that the Python API has changed; however, I struggle to find a board that causes this.

yaqwsx, 2023-06-27:
The situation is the same: I haven't found a way to reproduce this. We modified the command we show in GUI just to not confuse people in 27672b3. 

If you manage to supply precise steps to reproduce, feel free to reopen. At the moment, closing.

closed 2023-06-27 by yaqwsx