yaqwsx/KiKit #562: KiKit does not respect clearances in HV Designs

greenscreenflicker, 2023-06-26:
### Prerequisites

- [X] I have read [FAQ](https://yaqwsx.github.io/KiKit/latest/faq/)
- [X] I use KiKit at least version 1.3.0 (older version are not supported)

### KiKit version

kikit, version 1.3.0+26.g56c696a

### KiCAD version

7.0.5

### Operating system

Windows

### Description

I have a production ready PCB. However, when panalizing it, it get a lot of errors introduced and clearances are not respected.

Example Transformer original PCB
[image]

Panalized PCB
[image]

### Steps to Reproduce

@yaqwsx I send you all the files via email.

Execute kikit with the make gerbers command in the central folder.

yaqwsx, 2023-06-26:
I think this might be related to #536 - there were some changes in KiCAD that broke the net classes support.

greenscreenflicker, 2023-06-26:
do you think you can fix that in the near time?

yaqwsx, 2023-06-27:
This should be implemented in 119665a37c019fcaf5479f8e8c3be077688e6971 and f3ec40f94101a3315d2543135c162bb75392395e.

greenscreenflicker, 2023-06-27:
Checking with kikit, version 1.3.0+26.g56c696a, the error remains the same. 

I used the command >pip install git+https://github.com/yaqwsx/KiKit@master to update

yaqwsx, 2023-06-27:
Well, the current upstream is `1.3.0+35.g27672b3`. To update via pip, you have to add the `-U` flag.

greenscreenflicker, 2023-06-27:
Can you please add a note for the update in the docs? (bottom section?)
https://github.com/yaqwsx/KiKit/blob/master/docs/installation/upgrading.md

greenscreenflicker, 2023-06-27:
It seems that the upgrading does not work.
[image]

yaqwsx, 2023-07-23:
Could you try a clean reinstall of KiKit? Or invoke KiKit instead of `kikit` as `python -mkikit.ui`  I have no idea what's going on - clearly, KiKit +35 is installed, but launching KiKit launches an old version.

yaqwsx, 2023-09-14:
Closing due to inactivity.

closed 2023-09-14 by yaqwsx