yaqwsx/KiKit #563: KiKit does not ignore anual width when told so

greenscreenflicker, 2023-06-27:
### Prerequisites

- [X] I have read [FAQ](https://yaqwsx.github.io/KiKit/latest/faq/)
- [X] I use KiKit at least version 1.3.0 (older version are not supported)

### KiKit version

kikit, version 1.3.0+26.g56c696a

### KiCAD version

7.0.5

### Operating system

Windows

### Description

I disabled anual width warning, however KiKits brings up that error message.
[image]

[image]

### Steps to Reproduce

Make anual width issue and click on kicad to ignore it. export pcb with kikit and see that it brings up a warning.

You can use the PCB i send to you, should produce those errors.

greenscreenflicker, 2023-06-27:
btw: I don't know how to tell it to ignore padstack warnings.

Using the PCB i send you, there should be no errors remaining, however kikit raises some...

yaqwsx, 2023-07-24:
This should be fixed in 75e468e

closed 2023-07-24 by yaqwsx