yaqwsx/KiKit #567: Copper fill error when component have Edge.Cuts

cubesky, 2023-07-11:
### Prerequisites

- [X] I have read [FAQ](https://yaqwsx.github.io/KiKit/latest/faq/)
- [X] I use KiKit at least version 1.3.0 (older version are not supported)

### KiKit version

1.3.0

### KiCAD version

7.0.6

### Operating system

Windows 11

### Description

Copper fill with rotated component which have Edge.Cuts, have wrong copper fill.

[image]
[image]

### Steps to Reproduce

1. Make a component have Edge.Cuts inside.
2. Place it on the board.
3. rotate it to 30 deg.
4. Add copper fill
5. Panelized the PCB

yaqwsx, 2023-07-11:
Could you please provide the test board for steps to reproduce?

cubesky, 2023-07-11:
[EyeRGB.zip](https://github.com/yaqwsx/KiKit/files/12017642/EyeRGB.zip)

Here!

cubesky, 2023-07-11:
[image]

This is more bug for this, Edge.Cuts error. Non 90deg components with Edge.Cut missing Edge.Cuts.

cubesky, 2023-07-11:
[image]

JLC order system shown as no edge cuts

cubesky, 2023-07-12:
@yaqwsx I notice if millradius is 0mm. It shows wrong edge.

[image]

cubesky, 2023-07-12:
> [EyeRGB.zip](https://github.com/yaqwsx/KiKit/files/12017642/EyeRGB.zip)
> 
> Here!

Hi @yaqwsx,

This is my test board.

If you need more test file, please tell me!

cubesky, 2023-07-21:
Hello, is there any update for this？

yaqwsx, 2023-07-23:
This was fixed in 6e60d10.

closed 2023-07-23 by yaqwsx