yaqwsx/KiKit #570: Support for fileformat changes since 06/20/23

pferdob, 2023-07-21:
### Prerequisites

- [X] I have read the documentation and the proposed feature is not implemented.

### Description

Hi all,
sadly I started out with the 7.99 nightly months ago with a rather big project and above date the on-disk format changed slightly. I realized that much to late and wonder if there was any chance to get kikit working with the new file format as well. It will be part of v8.0 anyways...
Thanks in advance :-)

yaqwsx, 2023-07-23:
Hi, generally, we start to support a new major release in late November. Until then, there are numerous breaking changes every other week in KiCAD, and any effort in keeping KiKit compatible with nightly is basically wasted.

You can, of course, try to run KiKit with 7.99 (see https://yaqwsx.github.io/KiKit/v1.3/installation/choosing_kicad/), it might just work.

pferdob, 2023-07-23:
Thanks for your reply. I tried it - it does not work, because properties have been renamed. Will have to do it manually then...

closed 2023-09-14 by yaqwsx